iNavFlight/inav #7307: OSD CMS doesn't handle long lines well

MrD-RC, 2021-07-25:
## Current Behavior & Steps to Reproduce
There are a couple of options in the CMS menu where the lines are too long. This is not handled well by the menu system. This screen shot is from iNav 3.0.1 in the _Features > Navigation > RTH_ menu. Go to that menu to reproduce the issue. The top menu option should say **RTH ALT MODE** followed by the mode options. However, the mode options are so long that the spacing seems to clear all but two characters of the menu option title. There is also corruption of the menu options setting too. The option on screen should show **AT_LEAST_LINEAR_DESCENT**.

[image]

## Suggested solution(s)
I think the most simple solution would to be to allow a new line on menu options that are longer than the _total menu width - (the length of the menu options title + 1)_. Thought would have to be given that both lines would fit on the same screen.

Another option could be to crop the length of the menu options setting, so that the maximum length is _total menu width - (the length of the menu options title + 1)_. Though I think this option is not as attractive as the new line.

## Board
- FC Board name and vendor: Matek F405-STD
- INAV version string: Test firmware rebased to 3.0.1. The work is on an RTH function. Nothing on the OSD is different to a standard 3.0.1 release. https://github.com/MrD-RC/inav/tree/nav_rth_climb_first_stage

b14ckyy, 2024-03-20:
@MrD-RC was this fixed till now? Shouldn't be an issue for HD but for analog?

MrD-RC, 2024-03-20:
AFAIK it’s not been touched. I’m not sure it only affects analogue. I don’t believe the CMS uses the full with available with digital.