iNavFlight/inav #7319: Combine NAV_RTH_MAX_ALT and NAV_RTH_AT_LEAST_ALT modes

Pavel-DV, 2021-07-30:
## Current Behavior
The "nav_rth_alt_mode = MAX" mode is ignoring nav_rth_altitude value.

## Desired Behavior
There should be a mode that takes the maximum flight altitude, and compares it to the maximum predefined altitude, and uses the maximum one. 

## Suggested Solution
Add a new mode that is similar to the above mode, but does not ignore the nav_rth_altitude value.
For example, as a combination of two modes: NAV_RTH_MAX_ALT and NAV_RTH_AT_LEAST_ALT, when the highest possible altitude is used, for the greatest flight safety. 

## Who does this impact? Who is this for?
For those who fly at low altitude and sometimes fly around obstacles from the side, as well as in some cases fly around higher obstacles from above.

breadoven, 2021-07-30:
Easier to just change MAX so it respects the `nav_rth_altitude `value rather than have another RTH mode. If you want MAX to work down to takeoff altitude, just set `nav_rth_altitude `to 0.

domrachev-insegment, 2021-07-30:
> Easier to just change MAX so it respects the `nav_rth_altitude `value rather than have another RTH mode. If you want MAX to work down to takeoff altitude, just set `nav_rth_altitude `to 0.

I agree. This option would suit me. I would like the developers to hear us and make such an improvement. I think this is easy enough to implement.

Pavel-DV, 2021-07-31:
@breadoven 
I agree. This option would suit me. I would like the developers to hear us and make such an improvement. I think this is easy enough to implement.

MrD-RC, 2021-07-31:
This should work, but I need to test it.
https://github.com/MrD-RC/inav/tree/let-max-rth-mode-consider-altitude

avsaase, 2021-07-31:
@MrD-RC I'm not sure but isn't the home position offset already taken into account in the RTH altitude itself?

MrD-RC, 2021-08-01:
@avsaase it doesn't appear to be, based on the other RTH methods. If you look at the other methods, AT_LEAST for example; they all take the home altitude in to consideration.

PR #7447 (2022-01-09): Let MAX RTH mode consider RTH altitude
Fixes #7319 

MAX RTH mode now takes RTH Altitude into consideration.
- If RTH Altitude is 0, MAX behaves as the current mode. It will always return to the MAX altitude
- If RTH Altitude has a value
  - If the MAX altitude is lower than RTH Altitude, RTH Altitude is used
  - If the MAX altitude is above the RTH Altitude, MAX Altitude is used

Demo video https://youtu.be/YqSNMxzxiHM
closed 2022-01-09 by DzikuVx