iNavFlight/inav #7322: Rate profile selection on a switch

coulsonrobert, 2021-08-01:
## Current Behavior

I cannot make the adjustments tab change rate profiles based on an RC channel position.

## Desired Behavior

Setting an RC channel to a certain range gives a change in rate profile.

## Suggested Solution

Allow up to three rate profiles to be selected according to RC channel position.

## Who does this impact? Who is this for?

Many users. It would allow dual/triple rates or PID profiles on a switch. Useful for tuning too.

## Additional context

Some aircraft such as my full house glider have potentially different roll rates depending on whether full span ailerons are deployed. It is not always desirable to have full span ailerons active as they add to drag when thermalling. Tying the aileron/flap mode switch to separate rate profiles would be handy to allow a rate/PID profile for each mode. (Eg thermal mode with ailerons only and no rudder stabilisation, then aerobatic mode using full span ailerons and increased rudder P and I gains.)

padcom, 2021-08-09:
I am not entirely sure if I completely understand your problem, but it seems like what you're after is easily configured in OpenTX. There are flight modes which you can use and then configure the mixers as you wish.

> I cannot make the adjustments tab change rate profiles based on an RC channel position.

That is just weird. Are you stating that there is no option to choose P/I/D value change based on a channel or are you stating that you can't use the adjustments tab in the configurator for some reason?

coulsonrobert, 2021-08-09:
Essentially I would like two different PIDs and roll rates depending on the position of my flap lever switch. I don't think this could be solved using OpenTX as I want to change PID values not just rates.

Perhaps I'm doing it wrong, but if I use the adjustments tab to attempt to change a rateprofile based on an RC channel position it doesn't work. It seems to come up with some sort of error as my beeper sounds constantly but there's nothing showing on the OSD messages to shed any light on what is happening. The rateprofile does not change. Perhaps I"ll make a video of the behaviour and put a link here.

DzikuVx, 2021-08-23:
Duplicate of #5050

closed 2021-08-23 by DzikuVx