iNavFlight/inav #7326: Althold mode missing?

papage, 2021-08-02:
Hi all,

This is my first inav setup. I understand that there should be an ALTHOLD mode in the modes menu in the configurator, but there isn't! I have POSHOLD and many other modes but not ALTHOLD. Is this a bug or am i missing something?

I am running inav on a quad. 
- FC Board name and vendor: tyro129
- INAV version string:
-   CONFIGURATOR  3.0.1
-   FC FIRMWARE  3.0.1

Also, there is a minor bug with NAVPOSHOLD in the configurator. When I select from the TX the POSHOLD mode, the NAV POSHOLD should turn blue. Well it does not. It stays gray while the blue indicator (cursor) shows that the channel  value is in the blue range. This is a cosmetic bug of the configurator. The POSHOLD works, just fine. 
 
Thanx,
spiros

breadoven, 2021-08-03:
I assume you have a working barometer ? Althold mode won't be available if you don't unless INAV is set up to work with GPS altitude instead.

And the modes only show active in the Configurator modes tab if the conditions for them to activate are met, i.e. for Poshold you need to have a GPS fix.

papage, 2021-08-04:
yes, I have a barometer (tyro129) which works. The POSHOLD also works fine. 
When I fly I can see on the OSD, GPS coords, altitude (from the baro), compass etc.
The quad can also do ALTHOLD, if i go to CLI and issue a command like "aux 5 3 1 1350 1800" (3 is the ALTHOLD mode).
The problem is that ALTHOLD does not apperar in the configurator, in the modes tab, so it must be a configurator problem. I am not familiar with inav, so I am not sure if i need to do something in order for ALTHOLD mode to appear in the modes menu in inav configurator.

ElFab, 2021-08-04:
Is your baro red in the configurator? If yes try to power your board BEFORE connecting your USB cable. I must do that each time i need to go change something in the mode tab. Little borring to not be able to "show all"...

papage, 2021-08-14:
Yeap, that was it!
It is strange though that baro is red when i plugn the usb, because the tyro129 FC, is fully powered from the usb port, so the baro should work just fine...

Thanx!

closed 2021-08-14 by papage