iNavFlight/inav #7327: Problems of optical flow support

ZJyche, 2021-08-04:
I have a KAKUTE F7 running iNav 3.0.1.

At present, inav supports few optical flow devices
I am using PWM3901 optical flow(CX-OF) .I also have a gl-9306 optical flow.

In ardupilot, both gl-9306 and pwm3901 work normally
However, gl-9306 does not work properly in inav.

The data formats of both optical flows are : 
byte0: header (0xFE)
byte1: number of data bytes (0x04)
byte2: x-motion low byte
byte3: x-motion high byte
byte4: y-motion low byte
byte5: y-motion high byte
byte6: checksum
byte7: surface quality
byte8: footer (0xAA)

My range finder is vl53l0x.

closed 2021-08-05 by ZJyche

ralphwu, 2021-10-18:
> I have a KAKUTE F7 running iNav 3.0.1.
> 
> At present, inav supports few optical flow devices I am using PMW3901 optical flow(CX-OF) .I also have a gl-9306 optical flow.
> 
> In ardupilot, both gl-9306 and pmw3901 work normally However, gl-9306 does not work properly in inav.
> 
> The data formats of both optical flows are : byte0: header (0xFE) byte1: number of data bytes (0x04) byte2: x-motion low byte byte3: x-motion high byte byte4: y-motion low byte byte5: y-motion high byte byte6: checksum byte7: surface quality byte8: footer (0xAA)
> 
> My range finder is vl53l0x.

I have same issue on Omnibus F4 V3 FC.

mxchinist, 2023-09-19:
> I have a KAKUTE F7 running iNav 3.0.1.
>
> At present, inav supports few optical flow devices I am using PMW3901 optical flow(CX-OF) .I also have a gl-9306 optical flow.
>
> In ardupilot, both gl-9306 and [...]

How does the performance compare ? I also have both CX OF and gl9306, but the CX OF's accuracy falls apart pretty quickly in dim environment. Somewhere I read, that the gl9306 V3.0 should have improved performance in low light environment and on smooth/striped floors. Is that true ? Or is there just a minor difference ?