iNavFlight/inav #7329: Current version 3.1.0 is not compatible with inav configurator

Dave4675, 2021-08-05:
I tried compiling this version of inav but when I get into inav configurator i get the error that it can't configure it because is 3.1.0 instead of 3.0.1

stronnag, 2021-08-05:
3.1.0 is the development version, the release version is 3.0.1
If you really want a development configurator, you can either:
* Build your own from source
* Use the archives from http://seyrsnys.myzen.co.uk/inav-configurator-next/

closed 2021-08-05 by stronnag

derekquad, 2021-08-06:
i really need the IFLIGHT_SUCCEX_E_F4 /IFLIGHT_SUCCEX_E_F7 because i already purchase and was hopping to build  the hold position etc. is it possible? coz i dont the target in the inav configurator. but i saw the recommended  FC IFLIGHT_SUCCEX_E_F4

fanedi, 2021-08-13:
> i really need the IFLIGHT_SUCCEX_E_F4 /IFLIGHT_SUCCEX_E_F7 because i already purchase and was hopping to build the hold position etc. is it possible? coz i dont the target in the inav configurator. but i saw the recommended FC IFLIGHT_SUCCEX_E_F4

flywoo an hglrc have a lot of iNav compliant FC