iNavFlight/inav #7330: Foxeer F722 V2 Motor Output Error

GeckoLibre, 2021-08-06:
## Current Behavior
In the motor outputs tab, iNav target for Foxeer F722 V2 has an error where motor signal S2 does not spin the motor and causes the ESC to beep. Motor S1 does spin the motor, but it causes ESC beeps. Motor signals S3 and S4 seem to work fine. 

## Steps to Reproduce
1. Use any firmware version of FOXEERF722V2 or FOXEERF722DUAL with the Foxeer F722 V2 flight controller
2. After setting up a generic quad X mixer, go to the Outputs tab and attempt to spin all the motors.

Here is a video on YT showing the issue in more detail:
[https://youtu.be/9gM9n79_8cA](url)

- FC Board name and vendor: Foxeer F722 V2
- INAV version string: 
-
# version
# INAV/FOXEERF722V2 3.0.1 Jul  4 2021 / 13:46:18 (753c4479)
# GCC-9.3.1 20200408 (release)

DzikuVx, 2021-08-07:
Does the same happens with Multishot protocol? Care to attach a full config dump?

StrawHatSam92, 2021-08-07:
Hi Pawel,

I did not think to test multishot protocol unfortunately before I replaced the Foxeer FC with a Matek so I cannot confirm whether it works or not with multishot. I'm am not sure how to attach things in Github so here is the dump below:

`# version
# INAV/FOXEERF722V2 3.0.1 Jul  4 2021 / 13:46:18 (753c4479)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0
[... 1369 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 70
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 425
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH`

StrawHatSam92, 2021-08-14:
Hi Pawel, I reinstalled a different Foxeer F722 V2 onto my drone and the same problem persists, so I think that shows it's a firmware issue not a hardware issue. In addition, I took the time to see if the issue happens when using MULTISHOT. With MULTISHOT enabled, there seem to be no issues. Both when spinning up the motors in the configurator and when arming the motors with my radio (not connected to the PC), none of the motors shut down and it all works as expected.

In addition, I tried DSHOT300 outside of the configurator, just arming the quad with my radio and all four motors spin up, BUT as soon as I raise the throttle and give it any kind of stick input, the S2 motor shuts down and the S2 ESC reboots. So I don't think it's a configurator issue since it happens on the bench too. I will fly the quad anyway on MULTISHOT because I am also testing BlHeli 32.81 with variable PWM enabled (24-48kHz) to see how that goes.

prasanthsoman86, 2021-08-30:
I face the same issue with my F722 V2, and I quit Inav and flashed betaflight

PR #7410 (2021-08-31): Fix FoxeerF722V2 motor outputs
Fixes: #7330
Already tested by Foxeer
closed 2021-08-31 by DzikuVx