iNavFlight/inav #7338: [Security Issue] Stack-based Buffer Overflow

zidingz, 2021-08-09:
As per your [instruction](https://github.com/iNavFlight/inav/issues/7328#issuecomment-893732584), here are the details of the security issue:

# Description
There is a stack buffer overflow in the `stmloader` loader used by `inav`.
In `https://github.com/iNavFlight/inav/blob/master/support/stmloader/stmbootloader.c#L139-#L149` :
```c
char *stmHexLoader(serialStruct_t *s, FILE *fp) {
char hexByteCount[3], hexAddressLSB[5], hexRecordType[3], hexData[128];
char addressMSB[5];
static char addressJump[9];

// bzero(addressJump, sizeof(addressJump));
// bzero(addressMSB, sizeof(addressMSB));
memset(addressJump, 0, sizeof(addressJump));
memset(addressMSB, 0, sizeof(addressMSB));

while (fscanf(fp, ":%2s%4s%2s%s\n", hexByteCount, hexAddressLSB, hexRecordType, hexData) != EOF) { // vulnerable line
/**/
}
```
In the while loop, you call fscanf to read data from a bootloader file.

You use this format `:%2s%4s%2s%s` : the file must start with `:` then you copy 2 bytes in `hexByteCount`, 4 in `hexAddressLSB`, 2 in `hexRecordType` but you read an unlimited number of bytes from the file to the fixed size buffer `hexData`.
By loading a specially crafted bootloader, we can overflow that program and obtain arbitrary code execution.

# Proof of Concept
`echo $(python -c'print(":"+"A"*2+"B"*4+"C"*2+"D"*1337+"\n")')>evil_bootloader`
And run the bootloader using this file.

# Impact
Stack overflow, and possible code execution if successfully exploited

# Location
https://github.com/iNavFlight/inav/blob/master/support/stmloader/stmbootloader.c#L139-#L149

@digitalentity If you consider this a security issue, consider visiting https://huntr.dev/bounties/90eebb34-2aaa-4b94-a596-e19069267f71/ to validate the vulnerability, so that we can reward the security researcher. Let me know if you encounter any difficulties, or have any questions about our process. Thanks!

DzikuVx, 2021-08-09:
Irrelevant as the code is no longer part of the repository

closed 2021-08-09 by DzikuVx

zidingz, 2021-08-09:
I see, thanks.

In your opinion @DzikuVx, would this issue have been valid before the affected files were removed in https://github.com/iNavFlight/inav/pull/7334? Your take on this is important to us as the security researchers who originally reported this to us ought to be rewarded (from our side) for their work.

Thanks in advance for your cooperation!

DzikuVx, 2021-08-09:
No, it was not valid

zidingz, 2021-08-09:
Thank you for your response.