iNavFlight/inav #7343: new target for HOBBYWING-XROTORF7CONV

maxgame91, 2021-08-12:
## Desired Behavior
Is it possible to have a target for INAV for the Hobbywing F7 FC (name of the target in the title.
Can't find it in the confgurator

## Suggested solution(s)
An Inav target for this FC

## Additional context
dump : https://pastebin.com/TX20dVsz

- FC Board name and vendor: Hobbywing F7
- https://www.hobbywing.com/goods.php?id=681

ikherty, 2021-12-19:
it's actual for me and my friend too

ikherty, 2021-12-24:
[XROTORF7CONV_3.0.1_4.0.0.zip](https://github.com/iNavFlight/inav/files/7774637/XROTORF7CONV_3.0.1_4.0.0.zip)

closed 2023-10-27 by DzikuVx

AlejandroBaltazar, 2025-08-15:
@ikherty hi, by any chance do you have a more recent iNav HEX for the XRotor? The ZIP here has 3.0.1/4.0.0 only. If there’s an updated build or target/env name to compile myself, a link would help. Thanks!

ikherty, 2025-08-16:
> [@ikherty](https://github.com/ikherty) hi, by any chance do you have a more recent iNav HEX for the XRotor? The ZIP here has 3.0.1/4.0.0 only. If there’s an updated build or target/env name to compile myself, a link would help. Thanks!

 No, I don’t fly iNav anymore