iNavFlight/inav #7344: DJI OSD stats

mooiweertje, 2021-08-12:
## Current Behavior
Analog OSD shows stats when disarmed, DJI OSD does not.

## Desired Behavior
DJI OSD stats shows something similar to Analog OSD stats.

## Suggested Solution
In the fields that now show actual values, show stat values when disarmed.

## Who does this impact? Who is this for?
People who like to know some stats like max speed, altitude, distance from home, etc/

## Additional context
I added a few variabels to osd_dji_hd.c and some if statements  to make a quick and dirty implementation without reuse of what might already be available.

// stats when disarmed
static int maxSpeed = 0;
static float maxAltitude = 0;
static int maxDistance = 0;

[osd_dji_hd.zip](https://github.com/iNavFlight/inav/files/6975334/osd_dji_hd.zip)

Scavanger, 2021-08-31:
If you have suggestions for improvement, implement it cleanly and make a pull request, posting zip files here is useless. 
It's not worth putting any more work into it, Caddx has announced a canvas mode, so the complete OSD including statistics will be available soon.

mooiweertje, 2022-02-10:
Aaah what a pitty. It's such a nice feature.