iNavFlight/inav #7357: Auto throttle OSD over 100 value

jaroszmm, 2021-08-18:
During autolauch procedure in 3.0.1 I get OSD throttle values higher that set in configurator.
For 60% idle OSD shows 69 and for 100% launch I get 118.

nuts2351, 2021-08-25:
I had the same. autolauch throttle set to 2000, in OSD shows 120%.

b14ckyy, 2021-09-10:
I have the same issue on my X-UAV Clouds. And not only autolaunch is affected but also autonomous modes in general. 
Now while I am cutting a new video about the same plane I realized that also my Cruise throttle is always higher in the OSD than it should be. At perfectly level flight the cruise throttle is at 43-45% although it is set to 40% in INAV. So it seems all autothrottle values are boosted by around 15% in my case. And I have no idea why.

What ESC protocols do you guys use? Can you both please post a dump of your inav config?