iNavFlight/inav #7359: FR: External SPort Sensors injected into telemetry

wx4cb, 2021-08-19:
Not sure if this is implemented, is it possible to use a regular FRSKy Sport sensor, such as vario or even an arduino mimicing a sensor (eg: https://github.com/RealTadango/FrSky ) as an input on a serial port and send it out of the regular telemetry stream as a telemetry item?

i don't know if any processing would need to be done to it or whether it can be sent directly out as part of the telemetry stream.

Reason: I'm building a bunch of arduino based environmental measuring sensors and would rather be able to send it down as part of the telemetry as opposed to just storing it onboard and then waiting to see where i was when values i want....

for example, smoke particle measurements

DzikuVx, 2021-08-20:
@shellixyz isn't this the `smartport master` functionality?

wx4cb, 2021-08-20:
Im not sure that was why i was aski. Hopefully shellixyz can answer.

shellixyz, 2021-08-21:
It is possible to connect any sensor to an UART configured as "SPort master". It is emulating a receiver and allows to decode sensors' data to be used by INAV. Currently FLVSS data, vario data (altitude + vertical speed) and VS600 VTX data are decoded. These values are not currently used though. The data is also forwarded to S.Port telemetry so it is like you have them connected on the receiver. It also works the other way. S.Port commands are forwarded from receiver to S.Port master to be able to configure sensors/peripherals from the radio

wx4cb, 2021-08-22:
@shellixyz ok so if i have a sensor that's not one of those you listed, will it be passed through to the transmitter as a normal sensor?

Will the same work if im using crossfire? will it show up as an extra sensor ?

shellixyz, 2021-08-22:
The data sent by any sensor connected the to the SPort master port is forwarded. It is only forwarded to SPort telemetry on the receiver side though. It won't work with Crossfire

wx4cb, 2021-08-22:
> The data sent by any sensor connected the to the SPort master port is forwarded. It is only forwarded to SPort telemetry on the receiver side though. It won't work with Crossfire

bummer, so it wouldn't show up as a regular inav sensor ?

shellixyz, 2021-08-22:
No, only if you're using a receiver with S.Port/F.Port

closed 2021-08-23 by DzikuVx