iNavFlight/inav #7365: Update to INAV 3.0.0/3.0.1 breaks ghost duo receiver functionality

19Lukas, 2021-08-21:
## Steps to Reproduce

1.  Setup: Matek F722-WING, Ghost atto duo receiver on 1.0.4.2
2.  Everything works on Inav 2.6.1
3.  Update to 3.0.0 or 3.0.1 with configurator 3.1, receiver is not working anymore, no stick movements
4.  Go back to 2.6.1, works again

## Expected behavior
Receiver should work as on 2.6.1

## Suggested solution(s)

## Additional context
Only tested with ghost atto duo, not sure if the problem also exists with atto, although this should not make any difference I suppose
Inav 2.6.1 dump: https://pastebin.com/xH4Yv7g4
Inav 3.0.1 dump: https://pastebin.com/EQicuB3Y

---

- FC Board name and vendor: MatekF722-Wing  
Inav 2.6.1 version string: INAV/MATEKF722SE 2.6.1 Feb 25 2021 / 10:51:43 (64249b1a)
Inav 3.0.1 version string: INAV/MATEKF722SE 3.0.1 Jul  4 2021 / 14:31:07 (753c4479)
                                        GCC-9.3.1 20200408 (release)

Thank you!

DzikuVx, 2021-08-21:
Work fine over here with both Atto and Atto Duo. No problems. Rc works,
telemetry works. Latest firmware

19Lukas, 2021-08-21:
Thanks for your answer, any tip or idea what might be wrong?
I tried rebinding and also SRXL2, same problem, so I guess I have an incorrect setting somewhere.

DzikuVx, 2021-08-23:
Why don't you setup everything from scratch, since GHST is working just fine on all the boards I use

closed 2021-08-23 by DzikuVx

martink1974, 2021-09-16:
Don't think this should be closed I have same issue here with duo reciever on a speedybee f7 v2 ,if I put it in fast sbus mode it works but not in ghost or sxnl2 again going back to previous version 2.6 it works  , I don't see how it could be me as it works as soon as I rebind in fast sbus and change setting

19Lukas, 2021-09-16:
Try using a normal atto and then change it to the atto duo, that worked for me.

martink1974, 2021-09-16:
Don't think I have a spare normal here other than dismantling another
aircraft . It does seem very strange if that is point I would prolly have
to do that every update .