iNavFlight/inav #7366: vtx_halfduplex error, BrainFPV Radix Wing Kit, CRSF, SmartAudio

MileHighRy, 2021-08-22:
Here is the CLI status screen. I have erased and reflashed the FC and still having the same error. I tried to be able to SET it to ON or OFF and it would go away but then when it resets on closing CLI, it comes right back.  I have been searching for answers and not finding this VTX_HALFDUPLEX error anywhere else. Thanks for your help!

Setup:
BrainFPV Radix Li Wing Kit
CRSF Nano Diversity
TBS Unify Pro32 
iNav Configurator 2.6.1
FC Firmware 2.6.0

# status
System Uptime: 16 seconds
Current Time: 2021-08-22T16:26:01.503+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=180MHz, GYRO=BMI160, ACC=BMI160
STM32 system clocks:
  SYSCLK = 180 MHz
  HCLK   = 180 MHz
  PCLK1  = 45 MHz
  PCLK2  = 90 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x20020000, Heap available: 1628
I2C Errors: 0, config size: 6370, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 26, cycle time: 1243, PID rate: 804, RX rate: 148, System rate: 9
Arming disabled flags: CLI SETTINGFAIL
### ERROR: Invalid setting: vtx_halfduplex
VTX: band: R, chan: 5, power: 1 (25 mW), freq: 5806 MHz

# get vtx_halfduplex
vtx_halfduplex = ### ERROR: VALUE 5 OUT OF RANGE FOR vtx_halfduplex

Allowed values: OFF, ON

DzikuVx, 2021-08-23:
1. BrainFPV is not officially supported by INAV. For future problems you should contact BrainFPV customer support.
2. INAV tells you what the options are. `ON` and `OFF`.  Default is `ON` so please set it like that

closed 2021-08-23 by DzikuVx

MileHighRy, 2021-08-24:
Ok, thank you! My apologies, I didn't realize BrainFPV was not supported, their instructions state to use it. Thank you for the information!