iNavFlight/inav #7370: Extend Headtracking pan feature from home arrow to crosshair and include tilt

SilentZen, 2021-08-23:
## Current Behavior
crosshair does not move based on heatracking

## Desired Behavior
Have crosshair move down when looking up, move left when looking right and vice versa. Also have method to calibrate amount of movement.

## Who does this impact? Who is this for?
People who use headtrackers on front of planes and need assistance on identifying forward.

Jetrell, 2021-11-01:
If the Tilt servo is moved Up or Down more than 25°, the Crosshair will move off the screen, which defeats the whole purpose.

From my experience with head tracking; accelerometer drift requires the Pan and Tilt servo's to be re-centered many time per flight.

SilentZen, 2021-11-01:
I said have method to calibrate movement, you would have to set maximum
position based on servo movement!   Where do you get this magic 25% no from?

Jetrell, 2021-11-01:
`OSD_AHI_max_pitch`, can be used to keep the AHI within the bounds of the screen. 25° is a general number that allows the AHI to move from the Crosshair to the screen limit, with no text in the way. With text it will be less.  

I use the Pan home arrow offset feature.

> ## Who does this impact? Who is this for?
> 
> People who use headtrackers on front of planes and need assistance on identifying forward.

I can't say I've had an issue identifying forward from the Tilt offset. All my planes have a Tilt range around 120° (90° down 30° up). With no part of the plane in front of the camera. 
My Pan and Tilt have auto centering.

The only way this feature would work as you wish, is by having a non linear Crosshair position verses tilt servo travel.

SilentZen, 2021-11-01:
This impacts all plane pilots that mount their pan/tilt cameras on front of planes with no way of identifying centre forward.
This info is so important lot of pilots purposely mount pan/tilt cameras further back to see aircraft to identify centre position of camera.  The crosshair going off the screen is not an issue it can even disappear before edge of screen. With pan setups going way past cam FOV that would be expected, you cannot have centre forward been displayed when you are looking backwards.

closed 2022-02-10 by DzikuVx