iNavFlight/inav #7389: Corrupted Blackbox in OpenLog device

DavyJohnes, 2021-08-28:
## Current Behavior

Blackbox logs written to OpenLog device is not readable by iNav Blackbox Viewer

## Steps to Reproduce

1. Buy [this](https://aliexpress.ru/item/32838430172.html?spm=a2g39.orderlist.0.0.524a4aa6QMml6g&_ga=2.243842051.769031173.1629660517-997454257.1626388912&item_id=32838430172&sku_id=65107065542) device
2. Flash [this](https://github.com/cleanflight/blackbox-firmware) firmware
3. Flash iNav 3.0.1 to Matek F722 Mini se
4. Enable BLACKBOX feature
5. Select corresponding port as Blackbox on "Ports" tab
6. Connect batter and Arm your quad

## Expected behavior
Blackbox logs are readable by iNav Blackboox viewer

## Additional context
Example of blackbox file content: http://forum.rcdesign.ru/attachment.php?attachmentid=1584360&d=1629660978

- FC Board name and vendor: Matek F722 Mini se
- INAV version string: 3.0.1

stronnag, 2021-08-28:
There are many factors involved in logging via serial:

* The quality of the SD card
* The quality of the device 
* The logging rate

The best hardware is the OpenLager device e.g. https://uk.banggood.com/OpenLager-4-Bit-SD-Card-Data-Logger-Interface-Clocked-at-19_2MHz-with-DMA-for-RC-Quadcopter-Flight-Controller-p-1767137.html?cur_warehouse=CN which far exceeds the performance and reliability of other devices (e.g.legacy OpenLog).

For a start you might try lowering the logging rate, for example:

```
set blackbox_rate_denom = 32
# or even
set blackbox_rate_denom = 128
```
`blackbox_rate_denom = 32` works OK with an old AdaFruit Openlog, so there is no fault in inav,. It is more likely that your logging hardware cannot cope with the the date rate / volume (noting that that you have  `blackbox_rate_denom = 8`).

DavyJohnes, 2021-08-28:
> There are many factors involved in logging via serial:
>
> * The quality of the SD card
> * The quality of the device
> * The logging rate
>
> The best hardware is the OpenLager device e.g. [...]

Thanks fr your answer.
I user hight quality micr SD Card from Samsung with U1 speed type.

Ok, I will try to set `blackbox_rate_denom = 128`, previously it was 8.

avsaase, 2021-08-28:
FYI yesterday a change was merged to master that allows you to select which fields are logged, allowing for higher logging rates. With openlog it'll likely still be <<1kHz but at least higher than when all fields are logged.

DavyJohnes, 2021-09-06:
Yeah, that was logging rate, `blackbox_rate_denom = 32` works well. Thanks everyone!

closed 2021-09-06 by DavyJohnes