iNavFlight/inav #7403: Throttle Not Accurately Tracking

samtkeen, 2021-08-29:
Hello,
I am having this issue where my throttle in iNavFlight and my blackbox is not tracking properly. The throttle in the blackbox (RC Command) is tracking lower than what the reciever tab is recieving in iNav reciever tab. Originally iNavFlight recieved a range of 987-2011, i applied a 98% weight in the mixer to recieve a range of 1000-2000 for simplicity for the test. I also set idle power to 0% so we can actually see what 0 throttle tracks at in the blackbox

Equipment
FC - MatekF722 PX
firmware - iNav 3.0.1
Radio - FrSKY Taranis X9 Lite
Reciever - FrSky Archer RS
Is there a setting i need to change or something i need to do to make sure it tracks correctly?
if you need anymore details or questions please ask!

Attached is the blackbox recording

## Current Behavior
Test:
key: input in reciever tab of iNavFlight --> tracking of RC command throttle in blackbox
Arm to get base throttle -> tracks at 1000 (0%)
move to 1300/30% (hover throttle) --> tracks at 1196 (23%)
move to 1500/50% (mid throttle) --> tracks at 1377 (44%)
move to 2000/100% (max throttle) --> tracks at 1850 (100%)

## Expected behavior
The Inav reciever tab and the blackbox should track at identical positions

## Additional context
when switching to POSHOLD the RC command correctly tracks at hover throttle 1300 (30%)
[blackbox_log_2021-08-29_171923.zip](https://github.com/iNavFlight/inav/files/7072898/blackbox_log_2021-08-29_171923.zip)
[image]

stronnag, 2021-08-29:
Well, well, seems lile scaling works then.
```
$ grep --text throttle blackbox_log_2021-08-29_171923 
H minthrottle:1000
H maxthrottle:1850
```

samtkeen, 2021-08-29:
I attempted to change the max throttle to 2000 but the issue still persists.
Method CLI command: 
set max_throttle = 2000
save

kept the same 98% weighting on radio and it now tracks at:
30%/1300 --> 1222 (22%)
50%/1500 --> 1449 (45%)
100%/2000 --> 2000(100%)

so although max and min throttle track why is it all jumbled in the middle? NOTE all throttle expo's are set to 0

throttle params:
max_throttle = 2000
Allowed range: 1000 - 2000

throttle_scale =  1.000
Allowed range: 0 - 1

throttle_idle =  0.000
Allowed range: 0 - 30

failsafe_throttle = 1000
Allowed range: 1000 - 2000

failsafe_throttle_low_delay = 0
Allowed range: 0 - 300

fw_min_throttle_down_pitch = 0
Allowed range: 0 - 450

3d_deadband_throttle = 50
Allowed range: 0 - 200

airmode_throttle_threshold = 1300
Allowed range: 1000 - 2000

throttle_tilt_comp_str = 0
Allowed range: 0 - 100

samtkeen, 2021-08-29:
[blackbox_log_2021-08-29_185629.zip](https://github.com/iNavFlight/inav/files/7072989/blackbox_log_2021-08-29_185629.zip)

MrD-RC, 2021-09-21:
@stronnag shouldn't 1500uS (50%) scale to 1425uS?

I must admit, I'm not seeing the correlation, unless there is some sag compensation stuff happening?

I would expect:
| % | 1000-2000 | 1000 - 1850 | % of original range |
|---|------------|---------------|----------------------|
| 0% | 1000 | 1000 | 0% |
| 30% | 1300 | 1255 | 25.5% |
| 50% | 1500 | 1425 | 42.5% |
| 75% | 1750 | 1638 | 63.7% |
| 100% | 2000 | 1850 | 100% |

blind-oracle, 2022-01-12:
I'm having the same issue. The RC command in blackbox logs is always lower than I have on my TX.

In the attached blackbox log I have 1600us throttle set on my TX (no curves or whatever, same value received in INAV).
But in blackbox I see 1555us. I have `idle_throttle=0` and `max_throttle=2000` set, it was even worse before that.

[LOG00009.TXT](https://github.com/iNavFlight/inav/files/7853718/LOG00009.TXT)

DzikuVx, 2022-05-12:
Scaling works like intended. Throttle is also affected by `mincheck` parameter. That means that scaling is not simple [1000:2000]->[idle:max]

closed 2022-05-12 by DzikuVx