iNavFlight/inav #7417: smartport protocol sensor / telementry support

Outwide, 2021-09-01:
## Current Behavior
 Sensors Telemetry supported.

## Desired Behavior
 To be able to connect smartport protocol sensors and or Ibus as required.

## Suggested Solution

 To provide support so sensors can pass telemetry through FC to the receiver using Fport (smart and Sbus) and additional support for Ibus is also required.

## Who does this impact? Who is this for?
 This would be more in line Fc systems being compact and would better support users in technical fields. All users who want information on systems / Live Data.  

## Additional context
 Matek service has been contacted  and stated not any firmware(INAV,BF,ArduPilot) support this sensor over smartport protocol. It would be another  first for Inav.

MrD-RC, 2021-09-06:
There is SmartPort Master in the ports. That may do what you’re looking for? To be honest, I’ve not really played with that.

Outwide, 2021-09-07:
Hi Darren thanks for the email. My thoughts are to connect the FC with a port set to Smartport master  then to the TX for that UART to the in port on the back of the sensor, (piggyback) and then for the signal out connection to the F-Port on the receiver. My receiver has bother F-Port and Sbus, so I will try the F-Port. I note that I have seen the you can use on my receiver both F-Port and Sbus at the same time to the receiver, I don't think this correct. Will have a good look over the weekend coming. Will let you know if you like.