iNavFlight/inav #7418: Spline Curve to WayPoint navigation

JulioCesarMatias, 2021-09-03:
I don't know if anyone has thought about it to implement in iNav, but in the last few days I've been thinking about it. Adding Spline Curve mode to iNav would be nice, but I don't know if this is the right time for it. I have a sketch of the algorithm I got from ArduPilot, which is at the link below. Spline Curve mode is to make the UAV curves in wapypoint mode instead of straight lines, example in the images below. And I was also thinking that there should be a command in the CLI for the user to choose whether he wants Spline Curve mode active in waypoint mode or not.

https://github.com/JulioCesarMatias/WayPointSlipeCurve
[image]
[image]

closed 2021-11-15 by JulioCesarMatias

larrykabo, 2021-11-16:
needs to be an option

JulioCesarMatias, 2021-11-16:
> needs to be an option

it takes dedication to implement it in iNav and free time to test. And at the moment I don't have either