iNavFlight/inav #7422: Option to disable auto throttle for cruise mode.

Yarillo1, 2021-09-06:
## Current Behavior
I want to run my glider at very low throttle in order to have low amperage in cruise, But in cruise there is no direct throttle control, i can only increase throttle and my nav mode throttle value is too big for efficent flight. I could set throttle value for nav modes lower, but dont want to do so since it may be very bad in case of failsafe.

## Desired Behavior
I would like to manually set throttle in cruise mode.

## Suggested Solution
Add option to disable autothrottle in cruise. Or separate throttle values for rth and cruise.

## Who does this impact? Who is this for?
This is aimed on people who run long range planes in cruise.

## Additional context

MrD-RC, 2021-09-08:
To me, cruise is an automated flight mode. Therefore it should have auto throttle enabled.

For LR, the cruise throttle should be set at the most economic level to safely maintain flight. If the cruise throttle is not safe for RTH (Failsafe), then it is not safe for cruise.

a1axx, 2021-12-06:
You can use manual throttle override in cruise mode? So you can increase throttle to the set maximum or set minimum

closed 2024-03-20 by b14ckyy