iNavFlight/inav #7423: teadily losing the height and after some time it starts steadily rising of Drone

abhilash1998, 2021-09-06:
We built a small quadcopter using KAKUTE F7  and Matek 3901-L0X Optical Flow & Lidar Sensor without a compass for autonomous flight. When we put the drone in ALT HOLD it behaves weirdly and accelerates in a random direction and hits the net. When put in loiter, the drone drifts a little in the horizontal plane and after some time it starts steadily losing the height and after some time it starts steadily rising, but it is not an oscillation to consider it as a control problem. We found out that the altitude value from the matek sensor is pretty garbage. Hence we used TF mini as it is very accurate in altitude measurement, but it also has the same behavior.  We also tried tuning the PID parameters but it had no effect. After logging the data we observed the accelerometer values were pretty constant. The values of matek range-finder really fluctuated but for TF mini it was pretty constant. We tried tuning various parameters. The estimated value of height was varying. We reduced the weightage of the barometer and even nullified the values of the barometer. Another thing we tried was increasing the deadband but also it didn't work.
Can you please help me out solving this problem. Thank You
[LOG00006.TXT](https://github.com/iNavFlight/inav/files/7117762/LOG00006.TXT)

capblack, 2021-09-23:
I experienced a similar problem but with almost total loss of control

https://www.youtube.com/watch?v=iWu_CB-ViNo

DzikuVx, 2021-09-24:
[image]

The log shows that altitude follows target altitude no problems. However, rangefinder looses ground at around 1,5m of altitude

closed 2021-09-24 by DzikuVx