iNavFlight/inav #7426: I can’t fly with waypoint mode

mgphonexxxx, 2021-09-07:
Guys I need some help. My quadcopter was flash iNav3.0.1 and I work well . But I can’t fly Waypoint mode . I create mission in Mission Planner from iNav and save to FC . After that I load mission using stick command . When I load mission I heard 2 Beep ( “ Beep … Beep” ) . Then I arming my quadcopter and I switch to waypoint mode. Quadcopter fly 4 or 5 inches from ground automatically.. after that it land automatically. At my Goggle Screen ( Way Point Mission cancel and RTH ) . 

I don’t know what is wrong
Acc,Baro,Mag,GPS calibrate correctly .

Please Help me

breadoven, 2021-09-07:
You need to save to eeprom in Configurator Mission Control rather than save to FC. If you only save to FC and then power down/reboot the mission is lost and can't be loaded using the stick command. If you saved to FC and remained powered then the stick command will unload the mission (you don't need the stick command in this case since the mission is already loaded). If you end up in RTH when WP is selected it means there is no valid mission loaded.

mgphonexxxx, 2021-09-07:
Ok . i will try it. thanks for advice