iNavFlight/inav #7434: 3.0.1 FC freeze after a couple of minutes

RoadyFPV, 2021-09-10:
I'am tested the Current inav master (development build) from 2021-09-08.

After a couple of minutes the FC freeze (no servo reaction an no telemetry feedback)
RED and BLUE LED light up permanently

# diff all

# version
# INAV/MATEKF722SE 3.1.0 Sep  8 2021 / 12:09:32 (05629c86)
# GCC-10.3.0

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 2 0 100 0 -1
smix 1 3 0 100 0 -1
smix 2 4 1 50 0 -1
smix 3 4 2 -50 0 -1
smix 4 5 1 -50 0 -1
smix 5 5 2 -50 0 -1

# servo

# safehome

# logic

# gvar

# pid

# feature
feature -TX_PROF_SEL
feature -BLACKBOX
feature -AIRMODE
feature MOTOR_STOP
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI

# map

# serial
serial 2 2 115200 115200 0 115200
serial 3 2097152 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1700 2100
aux 1 1 0 1700 2100
aux 2 12 0 900 1300
aux 3 10 5 1700 2100
aux 4 36 4 1700 2100
aux 5 13 3 1700 2100
aux 6 15 9 900 1300

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 24 1 V
osd_layout 0 1 1 1 V
osd_layout 0 6 12 14 H
osd_layout 0 7 14 13 V
osd_layout 0 8 8 15 V
osd_layout 0 9 1 2 H
osd_layout 0 10 1 14 H
osd_layout 0 11 2 6 V
osd_layout 0 13 14 12 V
osd_layout 0 14 19 1 V
osd_layout 0 15 21 6 V
osd_layout 0 19 2 7 V
osd_layout 0 22 14 1 V
osd_layout 0 23 13 0 V
osd_layout 0 25 22 5 H
osd_layout 0 26 24 7 V
osd_layout 0 28 12 14 H
osd_layout 0 30 2 13 V
osd_layout 0 32 1 2 V
osd_layout 0 34 11 9 H
osd_layout 0 35 1 9 H
osd_layout 0 38 2 3 H
osd_layout 0 39 2 7 H
osd_layout 0 40 1 8 V
osd_layout 0 85 20 1 H
osd_layout 0 105 9 14 H
osd_layout 0 106 8 3 H
osd_layout 0 107 1 3 V

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_enabled = ON
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set acc_hardware = MPU6000
set acczero_x = 78
set acczero_y = 3
set acczero_z = -62
set accgain_x = 4083
set accgain_y = 4079
set accgain_z = 4013
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = BMP280
set rssi_channel = 7
set serialrx_provider = JETIEXBUS
set motor_pwm_rate = 4000
set motor_pwm_protocol = DSHOT300
set platform_type = AIRPLANE
set model_preview_type = 28
set small_angle = 180
set imu_acc_ignore_rate = 9
set imu_acc_ignore_slope = 5
set applied_defaults = 3
set gps_sbas_mode = AUTO
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 1500
set nav_rth_allow_landing = FS_ONLY
set nav_rth_altitude = 5000
set nav_fw_bank_angle = 45
set nav_fw_control_smoothness = 2

# profile
profile 1

set fw_p_pitch = 10
set fw_i_pitch = 15
set fw_ff_pitch = 70
set fw_i_roll = 8
set fw_ff_roll = 35
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set nav_fw_pos_z_p = 20
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

set throttle_idle =  5.000

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

stronnag, 2021-09-10:
On the basis that this is from http://seyrsnys.myzen.co.uk/inav_dev/, I've rebuilt the firmware with vendor compiler 10.20 (vice Arch's 10.3). Care to see if that helps?

breadoven, 2021-09-10:
Are you doing anything in particular when it locks ? Or does it happen just by powering it up and leaving it ?

RoadyFPV, 2021-09-10:
@stronnag OK i will try it

@breadoven It freeze when i power it up and leaving it and it freeze when i arm and move the sticks

RoadyFPV, 2021-09-10:
@stronnag OK first test looks good. 
I will continue to test and observe it. I will let you know if there are any complications

RoadyFPV, 2021-09-10:
FC freeze again. fc uptime around 15 min. i start a new test and log it with the transmitter.

RoadyFPV, 2021-09-11:
[image]
[image]

RoadyFPV, 2021-09-11:
picture one freeze around 27min

picture 2 freeze around 15 sec

one freeze without picture around 15min

RoadyFPV, 2021-09-11:
I cant find a reason why it freeze. but it look like that wit transmitter on the fc freeze. 

This Version has no freeze. Maybe its helpful?
INAV/MATEKF722SE 3.1.0 Jul 21 2021 / 12:24:28 (026621c)
GCC-9.3.1 20200408 (release)

stronnag, 2021-09-11:
As it's marginally easier to build (026621c) with gcc 10 then master with GCC 9, lets start there.
Can you please try the attached. 
[inav_3.1.0_MATEKF722SE_gcc10_026621c.hex.gz](https://github.com/iNavFlight/inav/files/7147895/inav_3.1.0_MATEKF722SE_gcc10_026621c.hex.gz)

Ah, having made the `diff` readable, I've just noticed `set serialrx_provider = JETIEXBUS` ... 
Perhaps the compiler is innocent.

RoadyFPV, 2021-09-11:
after a few tries, it really looks like the fc only freezes when the transmitter is turned on.

@stronnag ok i will try it

RoadyFPV, 2021-09-11:
[image]
Freeze again at 10 min

stronnag, 2021-09-11:
I was rather hoping that wasn't the result. Can you give the attached a test. I'm not sure I'll like the result.
[inav_3.1.0_MATEKF722SE_gcc9.3.hex.gz](https://github.com/iNavFlight/inav/files/7148198/inav_3.1.0_MATEKF722SE_gcc9.3.hex.gz)

RoadyFPV, 2021-09-11:
first run around 30min no problem, i will test more tomorrow.

stronnag, 2021-09-11:
> first run around 30min no problem, i will test more tomorrow.

I was afraid you were going to say that.

RoadyFPV, 2021-09-11:
That doesn't sound good at all. Is that a bigger problem?

stronnag, 2021-09-11:
> That doesn't sound good at all. Is that a bigger problem?

What we know so far:

* For your FC/RX combination:
  * anything compiled with GCC 9 works
  * anything compiled with GCC 10 fails

* For anyone else
  * We've no other reports of GCC 10 causing problems
  * I've using firmware with GCC 10 since 2020-05-16  (i.e for 15+ months) without problems

That's not to say that there is not some underlying "heisenbug"; if fact I suspect there may be. Examples like this issue, the recurring loss of configurations and the fact that no GCC 11 compiled firmware is usable seems more than mere coincidence.

OTOH, on my F4 / Frsky setups, GCC 10 compiled / current master is solidly reliable.

So "perhaps".   Maybe some people are unlucky with marginal hardware that the more optimised GCC10 firmware stresses.

RoadyFPV, 2021-09-12:
ok i have finished two another test. 

one around 25min and one around 50min. both with no problems.
Is that new hex usable for test flights, or better not.

[image]

stronnag, 2021-09-12:
> Is that new hex usable for test flights, or better not.

Only you can decide that. As I said, I have zero issues with the latest master and default compiler.

RoadyFPV, 2021-09-12:
> 
> 
> I was rather hoping that wasn't the result. Can you give the attached a test. I'm not sure I'll like the result.
> [inav_3.1.0_MATEKF722SE_gcc9.3.hex.gz](https://github.com/iNavFlight/inav/files/7148198/inav_3.1.0_MATEKF722SE_gcc9.3.hex.gz)

OK this file is compiled with the same compiler as this version INAV/MATEKF722SE 3.1.0 Jul 21 2021 / 12:24:28 (026621c)
GCC-9.3.1 20200408 (release), right? 

With the Version from jul 21 2021 i have done many test flights without issues.
Do you think there is a fix for this issue?

stronnag, 2021-09-12:
As the issue appears specific to your hardware, no.

RoadyFPV, 2022-01-26:
@DzikuVx  After I have now tried a lot, I think I have found the issue why the FC freezes randomly.

It really seems to be with the ExBus protocol and the GCC10 compiler. Tests showed that the version with GCC9 compiler does not lead to any freezes. When I use INAV 4.1.0 rc1 with ExBus the FC freezes involuntarily. 

Since the Jeti receivers can also output SBUS I switched to this protocol. The freezes are gone.
Unfortunately, the telemetry then no longer works.

This is a realy dangerous bug, that cost me a DartXL °RIP° and this should be followed up.

RobinatiNav, 2023-03-21:
I'm having the same problem. It cost me to crashes on EXBUS protocol. I use iNav 6.0 (GCC 10.x.x)

RoadyFPV, 2023-03-22:
Pls try to set the tx refresh rate to 50Hz.
Test on the bench.

RobinatiNav, 2023-03-22:
> Pls try to set the tx refresh rate to 50Hz.
> Test on the bench.

Thank you but I tried it couple weeks ago with no success.

RoadyFPV, 2023-03-22:
I have set it to 50hz and my freezes are gone. Same issue on betafight comiled now with gcc 10 too.

Issue is gone when i set it to 50hz tx refresh rate.

RobinatiNav, 2023-03-22:
> I have set it to 50hz and my freezes are gone. Same issue on betafight comiled now with gcc 10 too.
> 
> Issue is gone when i set it to 50hz tx refresh rate.

Lucky you, I crashed on 50Hz. What HW do you use? I have DC-16, RSAT2, MATEK H743-WLITE / DIATONE MAMBA mk4 F722 WING. Also, 45A ESC, 7 servos, LEDs strip, landing light, RSSI. I use 14 channels in this configuration and have some mixes configured on Tx side.

RoadyFPV, 2023-03-22:
DC24, Rex3, Matek F722 Wing, 2x 60A ESC, 4 Servos, DJI FPV Video TX, GPS

RoadyFPV, 2023-03-22:
If you turn off your tx the freezes should be gone

RobinatiNav, 2023-03-22:
> Contributor

> DC24, Rex3, Matek F722 Wing, 2x 60A ESC, 4 Servos, DJI FPV Video TX, GPS

I did not mention GPS and DJI O3 Air Unit

RobinatiNav, 2023-03-22:
> If you turn off your tx the freezes should be gone.
> Disable telemerry option should work too

It really funny. If I turn OFF my I cannot fly and what is the advantage of using EXBUS without telemetry?

RoadyFPV, 2023-03-22:
Thats true but it should explain that has something to to with the active connection to the rx and the telemetry option

RoadyFPV, 2023-03-22:
Definetly gcc10 broke something

RobinatiNav, 2023-03-22:
In my testing I always lost the telemetry on the Tx after some random time generally in some tens of seconds playing with a throttle. Fastest way to replicate it is to wait for prop ti stop and them move the throttle a bit and again and again until the telemetry is suddenly lost. When telemetry got frozen on Tx the FC still works, I can see it all telemetry data working in iNav a that time. It is veeeery hard to freeze the FC completely. I only faced that during the flight twice already.

RobinatiNav, 2023-03-22:
I am testing SBUS with ESC cables connected to FC with no freezes at all. Of course there is no telemetry but again the telemetry itself is not problem.

RobinatiNav, 2023-03-22:
The main reason why I investing my time to iNav is that I want to keep the model in particular altitude and horizontally stable. I use Jeti REX10A receiver. One of the flight modes is Horizon. In this mode only Roll movement is stabilised and pitch is set to fly slightly nose down. I don't like that mandatory descending. Loiter or Cruse mode would be perfect. I spent over 100€ for FC to only found out it is not reliable using EXBUS protocol which is supposed to be working 100%. What a shame. Now I have expensive OSD. Fortunately, I was able to configure my Jeti Tx to keep Altitude like I want even the Horizon mode tries to push model down.....

RobinatiNav, 2023-03-22:
> Compile INAV yourself (or ask a dev for it) with gcc9 and it works fine... I have the same issue and it annoys me tremendously
> 
> it should work with 100Hz refrash rate and telemetry as with gcc9

I don't know how to compile it. I have never done it.

RoadyFPV, 2023-03-22:
I have tried it so many times, its realy a pain in the a.... Iam using windows 

https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md