iNavFlight/inav #7449: SpeedyBee F7V2 Wifi BLE disable during ARM

abimaputra78, 2021-09-16:
@DzikuVx

Dear Pawel, 
Please kindly help for SpeedyBeeF7V2 board there was an internal wifi/ble chip. Can it be turned off during arm? To avoid 2.4Ghz interference. I requested this since in Inav no remap resources and pinio setup. 🙏  This is true PA15 control those enable like in Betaflight but I didn't see any correlation with arm flag. Not sure about this. 

Could u pls help. Thanks and appreciate this.

[image]

Thanks so much
Ditto

martink1974, 2021-09-16:
is this fixed is 3.0.2 ? this would explain failsafes i have had at medium  range on 3.0.1

martink1974, 2021-09-21:
no its not fixed this still affects range of 2.4g links

abimaputra78, 2021-09-21:
Yes. In 3.0.2 no pinio destination for turn it off. Pin A15

mrechte, 2022-07-09:
I confirm, ARMING does not disable Bluetooth module with iNAV 4.1.0.

Speedybee support says this should work out of the box with iNAV...

For BF they give some instructions on the CLI to make it work: https://speedybee.zendesk.com/hc/en-us/articles/5390784259099-Will-flashing-custom-firmware-for-SpeedyBee-F7-V2-flight-controller-affect-Bluetooth-and-WIFI-functions-

Does iNAV provides some equivalent commands ?

```
# resource
IO:
----------------------
...
A15: PINIO1 OUT
...

# get pinio_box
pinio_box1 = 47
Allowed range: 0 - 255

pinio_box2 = 255
Allowed range: 0 - 255

pinio_box3 = 255
Allowed range: 0 - 255

pinio_box4 = 255
Allowed range: 0 - 255
```

mrechte, 2022-07-10:
Could somebody kindly give directions in source code, where to customize target in order to change PÏNIO1 level when arming/disarming ?
Thanks

mrechte, 2022-07-20:
In Speedybee target.h, one can find:
   ` #define USE_PINIOBOX`

In main/fc/fc_core.c:processRX(), one can find:
```
#ifdef USE_PINIOBOX
    pinioBoxUpdate();
#endif
```

In main/io/piniobox.c:pinioBoxUpdate(), one can see a loop on the 4 pins (PINIO_COUNT):
    `pinioSet(i, IS_RC_MODE_ACTIVE(pinioBoxRuntimeConfig.boxId[i]));`

In main/fc/rc_modes.c, one can see:
```
bool IS_RC_MODE_ACTIVE(boxId_e boxId)
{
    return bitArrayGet(rcModeActivationMask.bits, boxId);
}
```

In main/io/piniobox.h, one can see
`#define BOX_PERMANENT_ID_USER1      47`

So I would assume that PINIO1_PIN is assigned to BOXUSER1 flight mode (defined in rc_modes.h) ?

In inav Mode config, I tried to assign USER1 to the same channel then ARM, expecting that it would drive PINIO1. Unfortunately it does not work.

closed 2023-10-27 by DzikuVx