iNavFlight/inav #7455: SmartAudio on softserial

jmajew, 2021-09-17:
HI,
I think I have found a problem with SA not working on softserial on Omnibus F4 Pro V3 (but I believe it can affect other targets as well). I have compared the results from logic analyzer between SA on hardware and software uart and the results are here:

Hardware UART (UART6 on Omnibus F4 Pro V3)
[image]

Software UART (S6 on Omnibus F4 Pro V3):
[image]

It looks almost right but the soft one starts the frame from high - and according to the TBS SA definition:
"The​ ​SmartAudio​ ​line​ ​need​ ​to​ ​be​ ​low​ ​before​ ​a​ ​frame​ ​is​ ​sent.​ ​If​ ​the​ ​host​ ​MCU​ ​can’t​ ​handle​ ​this​ ​it​ ​can​ ​be​ ​done​ ​by sending​ ​a​ ​0x00​ ​dummy​ ​byte​ ​in​ ​front​ ​of​ ​the​ ​actual​ ​frame."

I looked at the code and made a small fix (vtx_smartaudio.c function saSendFrame line 493):
was:
```
    switch (smartAudioSerialPort->identifier) {
        case SERIAL_PORT_SOFTSERIAL1:
        case SERIAL_PORT_SOFTSERIAL2:
            break;
        default:
            serialWrite(smartAudioSerialPort, 0x00); // Generate 1st start bit
            break;
    }
```
is:
```
	serialWrite(smartAudioSerialPort, 0x00); // Generate 1st start bit

```
and the result from LA (S6 on Omnibus F4 Pro V3)::
[image]
It seems that VTX starts to talk on softserial!
It would nice to check also other targets if they are affected or not. BTW, I think that betaflight has the same issue...

Jurek

---
- FC Board name and vendor:
Omnibus F4 pro V3

- INAV version string:
INAV/OMNIBUSF4V3_S6_SS 3.0.2 Sep 15 2021 / 10:33:19 (66011184)
GCC-9.3.1 20200408 (release)

MrD-RC, 2021-09-20:
If that change is in vtx_smartaudio.c it is target independent. Funny how it worked on some and not others. I didn't have problems with SA via SoftSerial on the Matek F411-WING/WSE for example.

Have you submitted a PR?

jmajew, 2021-09-20:
My guess is some VTX's don't follow to the letter the definition of the TBS SA. My luck that I tried to setup one that that do (Holybro Atlatl HV V2)...

I haven't submitted PR - I thought the changes are so tiny that someone during 'maintenance' can do it. Of course, I can create PR if necessary. Please, let me know.

MrD-RC, 2021-09-21:
@jmajew your PR could be that maintenance update 🤣 If you’ve done the work, you should make the submission.

closed 2021-10-16 by jmajew