iNavFlight/inav #7469: 1.3GHz (1G3) CMS Control via Tamp Protocol

Aerial812, 2021-09-22:
## Current Behavior
iNav is setup for easy UART control of 5.8Ghz video transmitters via IRC Tramp, but does not appear to allow control of 1.3Ghz VTX. 

## Desired Behavior
Allow users to connect 1.3G CMS control via Tamp protocol, or at a minimum setup a 1.3Ghz VTX using the CLI. 

## Suggested Solution
Currently the documentation on the Matex VTX provides details on how to set this up in betaflight. Those details are pasted below and available with more documentation at the link in the bottom of this request.

vtxtable bands 2
vtxtable channels 8
vtxtable band 1 1G3_A A 1080 1120 1160 1200 1240 1280 1320 1360
vtxtable band 2 1G3_B B 1080 1120 1160 1200 1258 1280 1320 1360
vtxtable powerlevels 3
vtxtable powervalues 25 200 800
vtxtable powerlabels 25 200 800

NOTE: only 1258 and 1280 can be enabled via CMS control on 2CH version.

## Who does this impact? Who is this for?
This impacts all users who want to be able to use 1.3GHz video transmitters and have the ability to turn the power up and down. It is currently a blocker to be able to use the leading hardware available for long-range FPV. 

## Additional context
The Matek VTX provides additional documentation on what is needed to support it.
Matek VTX-1G3SE: http://www.mateksys.com/?portfolio=vtx-1g3se#tab-id-2

wx4cb, 2021-09-23:
inav doesn't use vtx tables. also, IRC tramp support has been a part of inav for ages, so should work.

jeffhendrix, 2021-11-23:
MrD-RC, what's the latest status on this? I just ordered the matek vtx and I'm interested in this project. I was going to start working on it, but it sounds like you might have something close to working.

MrD-RC, 2021-11-23:
The code is done, I just need to test. I’ve been pretty busy recently, so not had the chance.

jeffhendrix, 2021-11-23:
If I get my vtx soon (it's on the Banggood slow boat), I'll test your code and post the results.

jeffhendrix, 2021-12-14:
MrD, got my VTX today. I thought I remembered seeing your changes somewhere, but can't seem to find them now. Do you have them posted somewhere?

jeffhendrix, 2021-12-14:
@MrD-RC , I just tested your code an it is working great. I used a Matek VTX-1G3SE 9CH VTX. Checked both bands and VTX properly changes channels (verified with LEDs on the board as well as transmitted video using RMRC VRX).
Also verified the power changes (with the onboard LEDs only). 
The only thing I couldn't get it to do was go into pit mode, but maybe there was something else I didn't have set right. Not being able to go into pit mode probably not a big deal for a plane (I would never use it).
I think you are ready to move forward with PR

MrD-RC, 2021-12-15:
Thanks @jeffhendrix I’m away with work. But as soon as I get the chance, I’ll make the PR.

MrD-RC, 2021-12-18:
Work has been an absolute nightmare at the moment. As this won’t be a 4.0 thing anyway. I’ll check out the pitmode thing before the PR. I don’t know anyone who has used pitmode with fixed wing. But, it should work if the spec is there.

Aerial812, 2022-03-09:
@MrD-RC & @jeffhendrix — Just curious if any progress has been made here. It sounds like the code is done, but maybe more testing is needed?

Fatih-Birkan, 2022-03-19:
@MrD-RC Can i test the code if possible ?  I started using these vtx's with INAV and controlling them from INAV would be great.
Thanks.

Jetrell, 2022-04-28:
@Aerial812 @jeffhendrix @rts18 @Fatih-Birkan @kealper @Orvo300
Would you people be so kind to leave a LIKE for the Darrens work on this feature, under the Pull Request link (above).. So its known to the head dev, that its a well requested feature.

Jetrell, 2022-04-28:
@Aerial812 @jeffhendrix @rts18 @Fatih-Birkan @kealper @Orvo300
Would you people be so kind to leave a LIKE for Darrens work on this feature, under the Pull Request link (above).. So its known to the head dev, that its a well requested feature.

closed 2022-05-07 by DzikuVx

MrD-RC, 2022-06-23:
No pitmode. Figuring out why that didn't work would need diving deeper in to Tramp than I understand at the moment. It should just be in the Tramp code that already exists. 

The general consensus was, who uses pitmode on 1.3GHz anyway. Most of the 1.3 VTx out there don't even have the option to change power.

PR #7949 (2022-05-07): Matek 1G3SE control via IRC Tramp
Fixes #7469 

Tested on F405-WING to MATEK1G3SE VTx hooked up to power meter.

- Verified that the channels change on the VTx LEDs and VRx
- Verified that the power changes on the VTx LEDs and power meter.
- The 5.8GHz side has been untouched, other than variable names.

[Demo](https://youtu.be/8JHr5VIYpCg)

To use this there are two things to note:
1. The CLI command `set vtx_frequency_group = FREQUENCYGROUP_1G3` must be used for this VTx
2. `vtx_channel` and `vtx_band` must be in a range accepted by the VTx. I will look in to a configurator change for this t [... cut]