iNavFlight/inav #7479: Resources not listed in CLI when calling dump

charlesmarseille, 2021-09-25:
Resources not listed in CLI when calling dump

## Current Behavior
dump does not contain resources list

## Steps to Reproduce
1.type dump in CLI

## Expected behavior
Resources be listed. They are not, I believe "resources" is called in the code, altough "resource" is the variable name.

## Suggested solution(s)
Modifiy "resources" to "resource" in dump command.

---
# INAV/MATEKF405 3.0.0 Jun 12 2021 / 12:25:44 (3c7b1b7c)
# GCC-9.3.1 20200408 (release)

stronnag, 2021-09-25:
If you want resources, you use the `resouce` command in the CLI; as they are not user-modifiable, it's pointless `dump`ing them.
An issue to remove the redundant `# resources` line from the `dump` would have more validity.

closed 2021-09-25 by stronnag

charlesmarseille, 2021-09-26:
Ok i think my case of wanting resources in dump is too specific.