iNavFlight/inav #7488: Flashing Firmware MacOS

mikekenyon99, 2021-09-30:
Can someone share the tips of how you flash boards with a Mac?  I have seem to spend my life getting the "No response from the boot loader".  I have tried with two Mac Mines and one MacBook Pro.  All of them are running the latest macOS Big Sur 11.6.   After about 30 mins of unplugging, restarting I eventually got one of my boards to update but now nothing. 

I have Matek F722-WING boards x2. 
It's the same on iNav Configurator 2.6, 3.0.0, 3.0.1, 3.0.2. 

What becomes a quick job turns into an evening epic every time.   What is the secret?

Mike

closed 2021-09-30 by mikekenyon99

Nanocastica, 2021-09-30:
The solution is easy but very anoyng, i  had the same problem and i have the same board,

Just unsolder all TX and RX ports from your board (if you have plug and play cables just disconect them), it is very bad solution but only one working, after that bootloader will work normally,

And if you want to change FC firmware you need to unsolder it all again or you need just to turn off all connected devices such as radio reciever GPS etc., cheers,

for me it worked when I unsoldered just battery cables of all my connected devices (radio and GPS),

Lets us know if it worked for you, but it should work,

mikekenyon99, 2021-09-30:
Feels a bit drastic.. however I always use servo pins so it wouldn’t be hard to try.