iNavFlight/inav #7557: If no barometer then modes depends on it should not be listed

mrnams, 2021-11-02:
## Current Behavior

## Steps to Reproduce

1. Connect FC to iNav Configurator which don't have barometer sensor or disable barometer from Configuration
2. Go to Modes in iNav Configurator
3. We will see Modes like NAV RTH,NAV POSHOLD, MAV WP etc which requires barometer

## Expected behavior

If mandatory hardware for mode is missing then those Modes should not be listed.
OR
Have option to Hide Modes whose mandatory hardware is missing
## Suggested solution(s)

Un-list Modes which are depend on some hardware and that hardware is missing.

## Additional context

---

# version
# INAV/MATEKH743 3.0.2 Sep 15 2021 / 10:30:36 (66011184)
# GCC-9.3.1 20200408 (release)
[image]
[image]

jeffhendrix, 2021-11-23:
Shouldn't this be an inav configurator request?

MrD-RC, 2021-11-23:
No, the firmware tells the configurator which boxmodes to display. So if the hardware is disabled, its down to the firmware not to send the modes over.

erstec, 2021-11-23:
@jeffhendrix
Currently all mentioned modes are active when it is Multirotor, have Compass, have Accelerometer and GPS.
Or if it is not Multirotor and have ACC and GPS.

There is no relations between them and BARO at all, except for NAV ALT HOLD and SURFACE.

Does anybody can confirm if mentioned modes should be really dependant on BARO? I don't see this relation from 'navigation' sources.
Then I will put PR to RC then.

erstec, 2021-11-25:
@jeffhendrix @MrD-RC 
According https://github.com/iNavFlight/inav/blob/master/docs/Navigation.md all those modes are non stricted to BARO and use it if present.

I think this issue can be closed.

closed 2022-05-12 by DzikuVx