iNavFlight/inav #7558: Caddx Peanut start/stop record function on a UART

rc-charlie, 2021-11-02:
## Current Behavior
The Caddx Peanut camera has been marketed "for FPV" as having a feature where you can start and stop recording via a UART TX pin. However, the only documentation describing how to go about this is for Betaflight (please see attached photo). Unfortunately (although I accept the reasons why... having seen Pavel's video on the subject) resource mapping is unavailable in iNav, so it is not currently possible to easily assign a UART TX pin for controlling the Caddx Peanut.

## Desired Behavior
The ability to assign a UART within iNav to start/stop recording of the Caddx Peanut camera.

## Suggested Solution
It is requested that the Devs please look into the possibility of adding a drop down menu item within the ports tab for any spare UART to enable control of the Caddx Peanut (similar to how there's a drop down item for DJI FPV for example).

Upon selection of this menu item, the selected UART TX pin has its serial function turned off, and instead it's reconfigured as a PINIO function set to an interval of 45,255,255,255 as per the attached image.

set pinio_box = 45,255,255,255

## Who does this impact? Who is this for?

This would greatly benefit all iNav users who wish to purchase the relatively new to market Caddx Peanut action camera. It is a relatively inexpensive camera compared to the GoPro series of cameras, it's very lightweight so good for sub250g builds and quads. I predict that many iNav users may wish to purchase this camera for their projects, but currently aren't able to take full advantage of it's remote start/stop feature.

[image]

donne1226, 2021-12-09:
I want this feature too

brousselle, 2022-01-03:
I have tried to enable pinio for my specific target MATEKF405SE as explained here: https://www.youtube.com/watch?v=2zElnakFQRU
It seems to work pretty well.
i left the default pinio_box1 at 47.

rc-charlie, 2022-01-03:
> I have tried to enable pinio for my specific target MATEKF405SE as explained here: https://www.youtube.com/watch?v=2zElnakFQRU It seems to work pretty well. i left the default pinio_box1 at 47.

Excellent, I'll go take a look.

sketch4x4, 2022-01-27:
Mine works fine following the instructions provided. One little issue is that you have to toggle the switch once to turn on the camera and then again to record. Which is different to the onboard button, where one press turns on and starts recording. 

Running this on a helio spring board 
resourced ledstrip to the pinio 1 (tx1 pad is damaged and I have no other uarts free), I used the same 40,255,255,255 provided in the manual.

AdamGFPV, 2022-03-16:
I'd love to have this feature, without the need to use the suggested workaround.

rc-charlie, 2022-03-18:
> Mine works fine following the instructions provided. One little issue is that you have to toggle the switch once to turn on the camera and then again to record. Which is different to the onboard [...]

You don't understand what is being asked here. Those instructions are just for betaflight firmware. this is an INAV git feature request, as resource mapping is not possible on INAV firmware

b14ckyy, 2024-03-20:
Implemented with https://github.com/iNavFlight/inav/pull/5993 on Flight controllers that support it.

closed 2024-03-20 by b14ckyy