iNavFlight/inav #7563: Velocity XYZ PID values not in diff/dump

stuwe911, 2021-11-05:
## Current Behavior
I adjusted my XYZ velocity values in the PID controller to get better automated flight. I was under the impression that a diff file would include all changes that are not default. After flashing and reconfigure i few FC's with the help of the diff/dump file i found that this information is not included. It always sets my PID controller back to default and i have to change it manually.

## Steps to Reproduce
1. open hidden PID settings
2. change values velocity XYZ
3. save
4. diff all / dump all
5. flash or clear FC
6. paste back diff/dump file
7. check PID settings velocity

## Expected behavior
This information to be included in a diff/dump file

## Suggested solution(s)
include velocity PID info in the diff/dump file

---
iNAV v3.0.2 and 4.0
Matek F722-WPX

rts18, 2022-01-04:
@StuweFPV  `Velocity_XYZ` do not appear to be tunable on a fixed wing.

[image]
.

StuweFPV, 2022-01-04:
thanks for your reply!
because i chose Airplane after flashing inav went into a kinda "airplane mode" and disables all MC codes hence no effect for the plane? usually it's noted in the configurator or the description but in this case it was not. however in that case it would make sense if they are not backed-up if they are not relevant. thank you for the clarification!

closed 2022-01-04 by StuweFPV

rts18, 2022-01-04:
I assume the values were hard coded originally, to provide an attainable window. 
Setting those values too high would set an impossible task for a fixed wing to achieve.  

Never the less, it certainly seems these values are a bit loose.  Most notably in 4.0.