iNavFlight/inav #7576: Odd behaviour of MSP Barometer

NameOfTheDragon, 2021-11-07:
## Current Behavior

There seems to be general weirdness around the handling of barometers that use MSP protocol.
I am using the Matek M8Q-CAN GPS/Mag/Baro unit.
I'm not using the UAVCAN port because iNav doesn't support that.
I'm not using the I2C port because my flgiht controller (ZEEZ F7 V1) doesn't have any I2C pads.
So I'm forced into using all three devices over a UART with MSP protocol, which works up to a point.

I configure my ports and devices like this:
[image]

[image]

[image]

In this configuration, GPS and MAG work fine, BARO is weird.

1. On power-up, iNav configurator shows that the baro is "OK" even if it is powered down or physically disconnected.
[image]

Once I plug in the battery, MAG and GPS light up too, but BARO is always there, even though it isn't.

2. If I plug in the USB first, then the battery, the barometer gives no readings (constant zero).
3. If I plug in the battery first, then USB, the BARO gives readings and I can see it changing on the Sensors screen.
[image]

4. Sometimes when I fly the quad, my OSD Altitude reading hovers around zero even if the quad is more than 50m in the air.

## Steps to Reproduce
1. You're probably going to need a BARO device that talks MSP, such as the Matek M8Q-CAN module with current firmware.

## Expected behavior
1. The barometer should not be indicated as "OK" when it is powered down or physically disconnected.
2. The barometer should work regardless of whether the USB or battery is connected first.
3. The barometer/OSD reading should not be stuck at zero when flying.

## Suggested solution(s)
I'm afraid I don't know enough about the internals of iNav to offer any useful suggestions.
I have tried to read the code and asked about it on both Facebook and Discord, but none of the devs have responded to my questions.

## Additional context

Config dump attached. [INAV_cli_Palantir_3_20211107_160858-iNav-Baro-Issue.txt](https://github.com/iNavFlight/inav/files/7492624/INAV_cli_Palantir_3_20211107_160858-iNav-Baro-Issue.txt)

- FC Board name and vendor: ZEEZ F7 (original V1 without baro or I2C)
- INAV version string:
> # version
> # INAV/ZEEZF7 3.0.2 Sep 15 2021 / 08:54:29 (66011184)
> # GCC-9.3.1 20200408 (release)

NameOfTheDragon, 2021-11-07:
I guess this answers point 1 - an MSP baro is always healthy (`barometer.c`).

```
bool baroIsHealthy(void)
{
    return true;
}
```

jaymeneto128, 2021-12-19:
@NameOfTheDragon I have the same issue. Using MATEK's F405STD. GPS and magnetometer not working under iNAV 2.6, 3.1 and 4.0 versions.
**QUESTION:**  Do you have GPS and / or magnetometer readings ?

NameOfTheDragon, 2022-01-07:
@jaymeneto128 
Yes, the GPS and Compass work fine, well even. The Baro works but not consistently, which is why I think it's an iNav issue, something to do with how it starts up that I can't pin down.

I don't know if this applies to your situation, but I had to flash the firmware on the M8Q-CAN module in order to get MSP support. The initial versions didn't have it enabled in the firmware, apparently. It took a lot of reading small print to discover that.

jaymeneto128, 2022-01-07:
Hi Tim Long, thanks for your reply,

The cause of my issue was using UART1 to receive MSP messages. My flight
controller is MATEK's F405-STD. Conversely, MATEK's M8Q-CAN only worked
when connected to UART 4 instead of UART1. I'm not sure if this odd
behavior is due to a defective port or from something related to INAV's
architecture.

We are witnessing a long period of rain here in Brazil, so I could not
perform an open field test yet in order to guarantee all is ok with my
setup.

THANKS - JAYME

robustini, 2022-06-01:
I have a similar problem and I'm going crazy for it.
Always Matek M8Q-CAN connected in MSP to UART5 on a Zeez F7.
The gps works, the magnetometer also works but the barometer remains in red.
Latest firmware on the gps installed, what could it be?

robustini, 2022-06-01:
@NameOfTheDragon actually the I2C is there, just edit the target, compile the code and it works.
But I wish it all worked in MSP.

[image]

robustini, 2022-06-02:
@digitalentity for safety I tried M8Q-CAN with Ardupilot connected to a Pixhawk, everything works, gps, magnetometer and barometer.
So please check your implementation if you have time because something is not working, thanks!
I also tried to change UART but it always does the same thing, the barometer is not detected.
Tested with iNav V4.1 and V5.

robustini, 2022-06-03:
> Hi Tim Long, thanks for your reply, The cause of my issue was using UART1 to receive MSP messages. My flight controller is MATEK's F405-STD. Conversely, MATEK's M8Q-CAN only worked when connected to [...]

But then did the barometer also work for you?
Today I also tried on UART1 and 2, same problem.
It is just something in the code that is not right.
That it worked on the developer's Matek is a coincidence, it should work on the other FCs as well.

robustini, 2022-06-03:
Damn I tried 4.1 again and now it works!
I hope the problem is fixed with the V5 finl because something broke the support for the brometer.

robustini, 2022-06-03:
@digitalentity damn, I tried 4.1 again and now it works!
I hope the problem is fixed with the V5 final because something broke the support for the brometer.

ricardodeazambuja, 2022-11-08:
I was having problems with the MSP2 barometer too, therefore I made some changes in my fork that make it work more like the other MSP2 sensors: https://github.com/thecognifly/inav/tree/1272353c35546c055dc5fd7448c5d6cbe6f9da9a

jaymeneto128, 2023-05-06:
Hi Ricardo de Azambuja, how are you?

About your patches:  I can still (safely) apply them under iNAV 6.0?  I
couldn't apply patches to "navigation_pos_estimator.c" because it changed a
lot since your last post.

THANKS IN ADVANCE - JAYME

ricardodeazambuja, 2023-05-06:
@jaymeneto128, my pull request (https://github.com/iNavFlight/inav/pull/8534) was merged into [6.1.0](https://github.com/iNavFlight/inav/tree/release_6.1.0).

jaymeneto128, 2023-05-06:
Ricardo, thanks for you response.

Can you give-me some help?

I've been using iNAV 6.0.0 since yesterday.
I tried to swap I2C <--------> UART4 pins from my MATEK's F405-STD  flight
controller because my controller's pad's are badly damaged. In order to
achieve my goal, I edited o arquivo ...\src\main\target\MATEKF405\target.h :

...

//    #define I2C1_SCL                PB6
//    #define I2C1_SDA                PB7
    #define I2C1_SCL                PA1
    #define I2C1_SDA                PA0
...

#define USE_UART4
// #define UART4_RX_PIN            PA1
// #define UART4_TX_PIN            PA0
#define UART4_RX_PIN            PB6
#define UART4_TX_PIN            PB7

....

In my build, TX4/RX4 are useless, so I tried to reuse these pads replacing
the original SDA/SCL pins. I compiled the target (with success), loaded the
HEX file into FC, checked if I2C assumed the new pinout with "resource"
command: all went OK.

However, no I2C signals are present in the new assigned PAD's. FC do not
recognize the BMP280 connected to the pads. More important: I checked with
my oscilloscope waveforms present and there is no clock and data flowing
(SDA) or clock present (SCL).

QUESTION:  What I'm forgetting?

THANKS IN ADVANCE - JAYME

b14ckyy, 2024-03-23:
I guess after the PRs above are all merged, this can be closed? If not, @ me

closed 2024-03-23 by b14ckyy