iNavFlight/inav #7583: H750/H730 CPU support

hydra, 2021-11-09:
H750 CPU support is required to support the SPRacingH7EXTREME, SPRacingH7NANO and SPRacingH7ZERO flight controllers.

http://seriouslypro.com/spracingh7extreme
http://seriouslypro.com/spracingh7zero
http://seriouslypro.com/spracingh7nano

The H750 CPU requires an external flash chip and is significantly cheaper than the H743 even when factoring in the external flash chip.

H730 CPU support is required to support the SPRacingH7RF and SPRacingH7EF flight controllers.

The following modes of operation are possible with the H750/H730 designs:

1) Copy firmware from SD card or external flash into RAM and run from RAM (fastest, easy to store config and supports logging).
2) Use a flash chip that supports memory mapped mode and run it from external flash (slower, more complicated to save config on the same chip, can't log at the same time to flash, can log to SD card).
3) Use two flash chips, one for code execution (small but fast), one for logging & config (large and/or cheap)

Note that the H730 CPU used in the H7RF (which also has an ExpressLRS capable RF chip connected via SPI) has the same requirements as the H750 and uses Option 2 from the above list.  The H7EF (unreleased, but BF source available on the SPRacing BF repo) uses option 3 from the above list.

The main tasks are as follows:

* Add H750/H730 CPU support to build system.
* Update build system to support generating EXST bootloader binaries
* Update H7 startup code (only minor changes to init code so that reboot reason handing is supported).
* Update MSP to allow rebooting to bootloader.
* External flash partitioning support.
* Add QSPI support for QSPI flash chip.
* Allow saving config to external flash.

Once that is done there are only minor changes between the H743 and H750 code.  H730 requires a little more work again because it uses OctoSPI instead of QuadSPI.

See also:
https://github.com/iNavFlight/inav/issues/6371#issuecomment-964117039
https://github.com/betaflight/betaflight/pull/10695

DzikuVx, 2021-11-10:
Awesome @hydra I would love to see your contribution to the project by making those changes and thus enabling pilots to use your hardware

hydra, 2021-11-13:
@DzikuVx I'm somewhat time-poor these days due to a new job - mainly posting the info above so that any interested party can tackle it.

If anyone wants to pick up this work I can provide hardware and technical assistance. Those interested should familiarize themselves with the support for the H7EXTREME in BF before tackling this job and contact me on the SP Racing discord server - https://discord.gg/99amK5K8ye

DanO83, 2022-01-07:
so is there any progress on the iNav target for the spracing H7 extreme?

hydra, 2022-05-09:
@DzikuVx It would be nice if there was a comment regarding the status of the ticket when closing it.  Such as 'no developer interest right now, reference in a new ticket if work is started', 'never'.   I'm guessing the former is applicable right?

DzikuVx, 2022-05-09:
Well, @hydra truth to be told, you have to answer this question. As we are happy to accept H750 support, you are probably the only one that has an interest in doing so. IIRC you are the only provider of H750 FC and it's up to you to do something about it.

As there is no movement in the topic, I have to assume you won't work on the topic, am I correct?

hydra, 2022-05-09:
@DzikuVx Someone contacted me today regarding getting started on this. I'm following up with them now and will help them get this done.  I don't have time to do it all myself or to 'own the code' as it were, but have time to assist technically as appropriate.

hydra, 2022-05-17:
@DzikuVx Can you re-open this, two developers have been shipped development boards today.

I've created an iNAV/SPRacing developers channel on the SP Racing discord server too, anyone wishing to help out is free to join the channel.

https://discord.gg/vP2sUGEhav

bkleiner, 2022-05-17:
Re-opening, Hardware has been shipped to me and @erstec
Further assistance is provided by @hydra via discord, but lets try to track progress here so its transparent

DzikuVx, 2022-05-17:
Cool!

Tenns, 2022-09-06:
MSAA! : make spracing ambivalent again! 
would love some h750 love spread into the world; Can a basically incompetent person such as myself help for h750 implementation? if not, I would still love to see some H7 extreme targets appear in INAV, and make those FC even more powerful.

bkleiner, 2022-09-06:
There have been some (rather large) delays, but hardware has been received and
i still fully intend to come back to this.

DanO83, 2023-03-26:
With the release of inav 6 I was hoping for support for the spracing h7 extreme, but unfortunately no activity from @hydra or other developers. should I give up hope and ditch the fc?

hydra, 2023-03-27:
@DanO83 Yes, it's a bit disappointing that the two developers that were sent hardware have not contributed.  Perhaps give the guys in the SPRacing discord #inav-dev channel a prod?  `@hanfer` and `@ernis` were the devs on the discord server.

DzikuVx, 2024-03-23:
Hi guys. Another year passed and I would like to know if H750 support is something plausible of going to happen in the near future?
If yes, then fantastic, I'm eager to have this rolled out in the next major release. 
If not, I'd like to close this issue because it's open for 2,5 years already and I seen no progress. Plus, it still seems like this FC family is the only one on the market that utilizes H750.
Thanks

hydra, 2024-04-09:
@DzikuVx Just close it, re-open if work happens by the devs that were sent boards.

I in related news ST released the awesome 600Mhz H7R3/7S3 and H7R7/7S7 MCUs which are what they call 'bootflash' MCUs which take a similar approach to using external memory for running code from like the H730.

closed 2024-04-25 by DzikuVx