iNavFlight/inav #7595: VGOODRCF4

cryptictuna, 2021-11-13:
Hi guys, i have acquired several VGOODRCF4 based on stm32f405    , there is firmware targets in CLeanflight, with same name as beteflight as well, VGOODRVF4 (STM32F405)    can this be ported over to INAV? it has 6 uarts, and onboard baro as well. would be perfect in Inav..

closed 2022-02-10 by DzikuVx

cryptictuna, 2022-02-11:
Closed with no resolve or even comments MONTHS LATER. ., .. so awesome

Closed #7595<https://github.com/iNavFlight/inav/issues/7595>.

DzikuVx, 2022-02-11:
And how do you expected this works? I buy samples for my own money and develop a target in my spare time? 
Manufacturer is apparently not interested in having this board supported in INAV and nobody else offered to build a target file.