iNavFlight/inav #7596: Multirotor modes shown when fixed wing connected

breadoven, 2021-11-13:
FPV ANGLE MIX and SURFACE appear as multi rotor modes when a fixed wing is connected. I'm guessing this might be related to the mode definitions in INAV fc_msp_box.c perhaps more than a Configurator issue ?

closed 2021-11-17 by breadoven

MrD-RC, 2021-11-18:
Just an addition here. It looks like SURFACE can be used on anything with LiDAR. So the switching of this should be based on available hardware, not platform. I’ll move it in to the modifiers section in the configurator.

erstec, 2021-11-18:
> Just an addition here. It looks like SURFACE can be used on anything with LiDAR. So the switching of this should be based on available hardware, not platform. I’ll move it in to the modifiers section in the configurator.

Should I push new PR with removed if()?

breadoven, 2021-11-18:
As it stands it appears that SURFACE is precluded for FIXED_WING_LEGACY which really means it's only good for multicopters:

https://github.com/iNavFlight/inav/blob/23973cb1a2bc0ebf472cd66d4a6f79e19a876a84/src/main/navigation/navigation.c#L918
https://github.com/iNavFlight/inav/blob/23973cb1a2bc0ebf472cd66d4a6f79e19a876a84/src/main/flight/mixer.c#L154

MrD-RC, 2021-11-18:
I note in the comments it says “not supported for fixe wing yet”. So I’ll continue to move it in configurator for future-proofing. But it sounds like hiding it from fixed wing is the right thing to do at this time.

PR #7605 (2021-11-17): Multirotor modes removed from Planes
Fixes #7596