iNavFlight/inav #7599: JHEMCUF7DUAL support inav 4.0

spikey1973, 2021-11-13:
Hi, infollow up of #5002 (which is closed) i would like to request JHEMCUF7DUAL support. 

As many before (#5002) i feel this is a great low budget board to be supported by inav. 
for now i will try to use the 3.0.0 file from the former topic, but if anyone is willing to make 4.0 available for tbis board it would be greatly appreciated. 

Kind greatz

Mattthieu

olegshty, 2021-11-15:
> Hi, infollow up of #5002 (which is closed) i would like to request JHEMCUF7DUAL support.
>
> As many before (#5002) i feel this is a great low budget board to be supported by inav. for now i will try [...]

I need it too! I've just build a wing and load hex of 3.0 from previus thread, but it can work  with only one servo because resoursces are not right... There are only 3 resources for 1-3 motors, but it's not aviliable in inav to make 2 motor out to work as servo((((
But this FC is perfect for wings and planes because it's cheap, has more than enought motor outputs and uarts!!!

jeffclassy, 2021-12-13:
I bought this FC too without checking whether there's a hex for Inav. I just looked at an FC with scl/sda pads. I guess it will take some time before I finish my LR drone.

jeffclassy, 2021-12-13:
Don't know much about making a hex file, but when I checked in the Betaflight codes, Flywoof7dual is close to the jhef7dual.
Some pins are switched out but I think it's do-able to just swap values. Reason why I chose FlywooF7 is because Inav4.0 have hex for it. 

https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/docs/boards/Board%20-%20FLYWOOF7DUAL.md
https://github.com/betaflight/betaflight/blob/025ee87a7aca068e3659fd066b8a9afbed123361/docs/boards/Board%20-%20JHEF7DUAL.md

terminat25, 2021-12-20:
[inav_4.0.0_JHEF7DUAL.zip](https://github.com/iNavFlight/inav/files/7742339/inav_4.0.0_JHEF7DUAL.zip)

flybabo, 2022-01-29:
I posted JHEF7DUAL 4.1.0-rc1 hexes [here](https://github.com/iNavFlight/inav/issues/5002#issuecomment-1024813954)

closed 2023-10-27 by DzikuVx