iNavFlight/inav #7604: Add 2 decimal places to Azimuth

jorgermp, 2021-11-14:
## Current Behavior
Azimuth is currently displayed in the OSD as an integer.

## Desired Behavior
For long-range flights, it would be nice to have 2 extra decimal places for the Azimuth and to be able to adjust the direction better.

## Suggested Solution
I don't know how to do this in code.

## Who does this impact? Who is this for?
Long range users for airplanes or fixed wings.

erstec, 2021-11-15:
@jorgermp it will requires a lot of changes in datatypes passed and to functions processed them.

@DzikuVx is this required?

DzikuVx, 2021-11-15:
To be honest I'm not sure how much directional the antenna would have to be so that decimal part matters. This sounds like an overkill requirement if you ask me

closed 2021-11-19 by DzikuVx