iNavFlight/inav #7608: Version numbering: 4.0.0-rc1 just after 3.0.1

soldier791, 2021-11-16:
Hi!
I see 4.0.0-rc1 version tag just after 3.0.1

1 Will there be versions 3.1, 3.2, etc?
2 What is the point of changing versions so quickly?

I am frightened by such a quick version change, and I start to think that the project has a lot of problems.

MrD-RC, 2021-11-16:
The version number selection method has changed. That is all. iNav now uses the SemVer numbering methodology. The major version changes if there is something that breaks backwards compatibility with the previous version. In this case, the fonts used on the OSD will not work with 3.0 or vice versa. Major versions changing is now not reliant on features, but compatibility. 

What is now 4.0 was 3.1 earlier in the development phase. There is nothing to worry about.

closed 2021-11-16 by DzikuVx

soldier791, 2021-11-16:
> The major version changes if there is something that breaks backwards compatibility with the previous version.

That is, version 4.x will be incompatible with 3.x?

MrD-RC, 2021-11-16:
> That is, version 4.x will be incompatible with 3.x?

Yes. But this has happened many times in the past. For instance the changing of the mode IDs between versions 2.4 and 2.5. That broke backwards compatibility. Using SemVer, that would have forced the major version to be bumped to 3.0. These changes that break compatibility don’t need to be massive things, nor even impact the flight characteristics. It really should make no difference to how people use iNav either. As you should have always used the same major and minor version of the configurator and firmware.

soldier791, 2021-11-16:
>Yes. But this has happened many times in the past.

Thanks you, Darren. 

I was not so much worried about the change of numbering as that it was not available annotated. 
This project is important for me and, I'm sure, everyone in the flying hobby. 
 
Thanks again.