iNavFlight/inav #7622: PosHold with Optical flow not working above 1m height

andrefreitas97, 2021-11-22:
## Current Behavior
When using PosHold indoors with Matek lidar 3901-L0X (and ALTHOLD with barometer) the quad keeps horizontal position hold only below a certain altitude (~1m). Above that height the opflow sensor seems to lose reference and the quad starts to drift. I already maxed the values of the cli commands nav_max_terrain_follow_alt and inav_max_surface_altitude and the problem still occurs.

## Expected behavior
Since the opflow sensor PMW3901 in the matek board should work between 80 mm and infinity, I expected that position hold should work above the  80 mm height to the max value defined in the cli.

## Suggested solution(s)
I am not sure how this can be solved.

# version
# INAV/KAKUTEF7HDV 4.0.0 Nov 12 2021 / 10:00:08 (a6c7d81c)
# GCC-10.2.1 20201103 (release)

FC Board name and vendor: Kakute F7 HDV

DzikuVx, 2021-11-25:
Used lidar has a practical range of 1m. Hardware limitation, won't fix

closed 2021-11-25 by DzikuVx

andrefreitas97, 2021-11-25:
My problem is related to the range of the opflow sensor and not the lidar. What I don´t understand is why the opflow is not capable of keeping horizontal position hold above 1m (**not using surface mode**. only althold with barometer and poshold)

avsaase, 2021-11-26:
The opflow sensor cannot properly see the ground when the distance is too large.

andrefreitas97, 2021-11-26:
Thanks for the reply avsaase! I already expected that could be the problem but what I don´t understand is why they advertise that this sensor has a range of 80cm to infinity when it actually "stops working" at about 1m of the ground.

DzikuVx, 2021-11-26:
No no. The OpFlow sensor has a range up to infinity. But it only return angular translation. To convert that to linear translation distance to tracked surface is required. And this comes from lidar. When you go out of lidar range, opflow data becomes useless

andrefreitas97, 2021-11-26:
Oh!! Finally someone can perfectly explain the reason behind it! Thank you so much!

VanLaser, 2022-04-27:
> No no. The OpFlow sensor has a range up to infinity. But it only return angular translation. To convert that to linear translation distance to tracked surface is required. And this comes from lidar. When you go out of lidar range, opflow data becomes useless

I have a related question - is it possible to add another lidar (like TFMini Plus for example), and use it in conjunction with the opflow data from the Matek unit? Basically to achieve what the OP asked. Thanks for all the INAV work BTW!

andrefreitas97, 2022-04-27:
Yes it works! I did exactly that and know opflow works until the max altitude of the tfmini

VanLaser, 2022-04-27:
That's great, thanks! :)