iNavFlight/inav #7623: Led Strip on H7 not working no output signal on led pin out - Tested on MATEKH743-Mini

Cal1sto, 2021-11-22:
## Current Behavior
Led Strip function set on PA8 doesn't send out any signal to activate and set color on selected led number (in CLI resource seems mapped correctly to A08, features set for led strip, and led set and controled in CLI)
Led strip type WS2812B, checked with an oscilloscope no signal on the led pin out to manage the led strip.

## Suggested solution(s)
A short exchange on Discord with Hanger suppose a DMA issue.

## Additional context
# feature
Enabled: VBAT GPS TELEMETRY CURRENT_METER LED_STRIP BLACKBOX AIRMODE PWM_OUTPUT_ENABLE OSD 
# led
led 0 0,0::C:2
led 1 1,0::C:5
led 2 2,0::C:9
led 3 3,0::C:1
led 4 0,0::C:0
...

# resource
IO:
----------------------
A00: MOTOR3 OUT
A01: MOTOR4 OUT
A02: FREE 
A03: FREE 
A04: FREE 
A05: SPI1 SCK
A06: SPI1 MISO
A07: FREE 
A08: LED STRIP OUT

# version
# INAV/MATEKH743 5.0.0 Nov 15 2021 / 22:59:35 (10084ebf)
# GCC-10.2.1 20201103 (release

- FC Board name and vendor: MATEKH743-MINI
- INAV version string: 5.0.0

Cal1sto, 2021-11-22:
On INAV default setting the led strip is running , but soon the mmix is set like following:
mmix 0  1.000  1.000  1.000  1.000
mmix 1  1.000  1.000 -1.000 -1.000
mmix 2  1.000 -1.000  1.000 -1.000
mmix 3  1.000 -1.000 -1.000  1.000

This modify somehow the buffer content where are stored the led strip parameter, making strange led output or not displaying the setting like it is configured in led setting.

Cal1sto, 2021-11-22:
Also loading and applying  a Mixer will generate same issue.

Cal1sto, 2021-11-22:
When i delete the Motor Mixer rule the led strip is working again

Cal1sto, 2021-11-22:
Setting only first motor with : mmix 0  1.000  1.000  1.000  1.000 is enought to generate the issue. (S1 | PB0 | TIM3_CH3)

mrnams, 2021-11-24:
@Cal1sto I have Matek H743 and LED works fine for me

Cal1sto, 2021-11-24:
@mrnams For me is working only without mixer rule on my quad, soon a mixer is set it is no more working

bkleiner, 2021-11-27:
Led indeed seems to stop working once a mixer is setup ***and dshot motor output is enabled*** which further hints at an dma issue.

PR #7633 (2021-12-01): h7: fix led strip dma configuration
fixes #7623
data coherency issue of `ledStripDMABuffer` on h7 by moving it to `DMA_RAM`
assigns LED pin PA8 a dedicated DMA_TAG to avoid conflict with S1 when its configured as dshot
closed 2021-12-01 by bkleiner