iNavFlight/inav #7626: OSD horizontal position

IvanBiv, 2021-11-23:
## Behavior

The crosshair symbol and home arrow are not at the same horizontal position, but inav configurator is fine.  A similar problem was previously here: https://github.com/iNavFlight/inav/issues/4911

## Additional context

[image]
[image]

---

- FC Board name and vendor: matek f411-wse
- INAV version string: 3

erstec, 2021-11-23:
@IvanBiv 
Can you provide your dump?

I don't see anything wrong, but will try with your configuration

[image]
[image]

IvanBiv, 2021-11-23:
# dump

# version
# INAV/MATEKF411SE 3.0.2 Sep 15 2021 / 10:13:43 (66011184)
# GCC-9.3.1 20200408 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 1 100 0 -1
smix 1 4 0 100 0 -1
smix 2 5 2 100 0 -1

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# logic
logic 0 0 -1 0 0 0 0 0 0
logic 1 0 -1 0 0 0 0 0 0
logic 2 0 -1 0 0 0 0 0 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 -1 0 0 0 0 0 0
logic 5 0 -1 0 0 0 0 0 0
logic 6 0 -1 0 0 0 0 0 0
logic 7 0 -1 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0
logic 16 0 -1 0 0 0 0 0 0
[... 1380 lines cut ...]
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 425
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 300
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# end the command batch
batch end

MrD-RC, 2021-11-23:
The configurator and OSD never lined up since the OSD positioning was changed in 2.6. This is fine in 4.0. I checked the elements when the new fonts were checked.

closed 2021-11-23 by MrD-RC