iNavFlight/inav #7629: RTH Autoland in a straight line with takeoff heading

FelixBaum310, 2021-11-25:
## Current Behavior
Landing in a circle is not useful at any kind of airfield

## Desired Behavior
Landing in a straight line with touchdown at the homepoint

## Suggested Solution
Average over the first x seconds of launch heading, remember it, interpolate a point at rth altitude height, opposite to launch heading behind the launch point with the slope being x% of the maximum sink rate. Then "just" rth to that waypoint and fly a straight line down towards the launchpoint

## Who does this impact? Who is this for?
fixed wing airplane users

## Additional context
none

MorrisR2, 2022-01-01:
This issue has been raised several times.  There is a pull request to implement it.

I will say I disagree strongly with "Landing in a circle is not useful at any kind of airfield".
The current behavior is useful anywhere that the area free of trees is roughly as wide as it is long - soccer fields, etc.
A straight-in approach is needed for landing on a dirt road, where there are trees on either side but not at either end.

StuweFPV, 2022-01-04:
I could also do with that feature. PitLab does a fantastic job with full ILS landing exactly the way you want

wx4cb, 2022-01-07:
also dont forget.. you're taking off into the wind so you can't come in from the opposite direction and be landing with the wind. that's just asking for trouble

cfitches, 2023-03-08:
> also dont forget.. you're taking off into the wind so you can't come in from the opposite direction and be landing with the wind. that's just asking for trouble

Yes this is asking for trouble. But arguably asking for less trouble than circling in a lot of circumstances. We don't really care about the plane, we just dont want it to hit something.

b14ckyy, 2024-03-20:
Done https://github.com/iNavFlight/inav/pull/9713

closed 2024-03-20 by b14ckyy