iNavFlight/inav #7635: Add a time dimension to the logic/programming features

NameOfTheDragon, 2021-11-27:
## Current Behavior
Currently, there doesn't seem to be a way to get time-dependent behaviour in the logic/programming system.
This would be useful because there is a need to detect persistent conditions, timeouts, etc. which needs a timing mechanism.

## Desired Behavior
Introduce the concept of a TIMER to the logic system.
- Timer has two inputs:
1. A logic condition that starts/stops the timer (the trigger)
2. A timeout interval (expressed in milliseconds)
- Timer has one output:
1. A logic condition that is true if the timer has timed out, false otherwise.
- Timer is triggered (started) when its input logic condition becomes true. If the condition goes false, timer is reset and stopped.
this could be implemented as two separate logic inputs (one for start, another for reset) but, all else being equal, I think the single input approach is cleaner.
- If the timer counts up to the timeout value, its output becomes true and remains true until the timer is reset by its trigger becoming false.

## Who does this impact? Who is this for?
This is for all users. It will generally improve the programming capabilities by adding a time dimension, which will be really useful for detecting timeouts, persistent conditions, etc.

The main use-case I have in mind is a simple crash detection logic. If the craft is inverted (-90 <= roll <= +90) for more than "a few seconds" then it will be disarmed. A few users (on Facebook) have reported their craft crashing inverted and then trying to drill into the ground with no response from the disarm switch. This could happen for example if the receiver holds all channels at the last known good value when it detects a failsafe condition, as some FrSky receivers might do. All of the features (except a time delay) are available in the logic/programming screen to achieve this.

ABLomas, 2021-11-27:
Timers are widely used in sequencers. For example, scale model:
 - open landing gear door
 - wait 2 secs for operation to complete
 - drop landing gear
 - wait 5 secs
 - move door to fix landing gear
 - wait... 
For effects, for lightning, for many purposes - but usually done on TX.
Having timer in INAV programming would allow to move all sequencer logic into FC (this way even crude TX (do they still exists today? :D ) can do all stuff, or allow sequences in autonomous/failsafe flight).

Jetrell, 2021-12-24:
#6447

closed 2023-03-17 by MrD-RC