iNavFlight/inav #7637: SpeedyBeeF7V2 beeper and dshot-beacon always on

Bigboostfpv, 2021-11-29:
The problem affects every available version of INAV, even 4.0. BF does not have the problem. "beeper -all" is ignored.
"set dshot_beeper_enabled = off" turns it off completely also the mode.
It is only on dshot not MS. It turns off if i connected to blheli or when I give something throttle in outputs.

piwi3910, 2021-12-12:
i have the same board, running inav 3.0.2, no issues with the beeper.
beeper will be on if armed, and no throttle given. you can set the same setting in BF.

Joerg-H-Wagner, 2021-12-22:
Reason is a warning in general. DSHOT is able to beep so you hear the beep like a beeper. 
An example with my quad with SpeedyBeeF7V2 and GPS module Beitian BN-880:
The setting of magnetometer in CLI was set to HMC5883 but it has to be set to QMC5883.
After solving this hardware issue the beeping was gone and I can use DSHOT600. 
Please check your hardware settings in detail. Good luck.

ddave47, 2021-12-27:
Running SpeedyBeeF7V2 and iFlight M8Q-5883-GPS Module V2.0.  INAV 4.0.0. Magnetometer set to QMC5883. 
I ran into what I believe to be beeper problems when I tried to set my throttle esc endpoints by using the motor test mode. The escs would not go into endpoint setting mode.  Had to set esc protocol to STANDARD, then set my endpoints, then reset protocol to dshot600 to get them set. Not sure if this is a bug or not.