iNavFlight/inav #7638: Support for CRAZYBEEF4SX1280 with ELRS SPI

RipperGOT, 2021-11-30:
Hello friends, would it be possible to support this controller? Currently only Betaflight supports it. Yes, beta is fine, but I love INav!😍
This is the page where they give you support. 

[https://github.com/betaflight/betaflight/pull/10788](url)

bishko, 2022-02-10:
I would also love to have INAV on this little guy. It's built for INAV because this is a slow cruiser made for short waypoint missions and simple flying, not for full acro race style flying! INAV would be perfect for this!

Sjullep, 2022-04-05:
I would like to have this too for the DarwinFPV Tiny Ape

mirov, 2022-05-13:
As strange as it sounds, I'm trying to use this controller for a small fixed-wing.  It offers great integration, a ELRS receiver, and a boost converter to give things 5V.  It would be nice if the boost would let me drain the cell down to ~2.4V (I'm using a LION, so that's OK).  It has a serial port based receiver, so it getting this to run with INAV as simple as a new target.c and target.h set of files ?

RipperGOT, 2022-07-03:
Well, looking at the problems for future updates with SPI, and the problems getting it to work right in Betaflight. I think we can close this.

closed 2022-07-03 by RipperGOT

mirov, 2022-07-03:
I have been using this FC with INAV for a bit now (fixed wing).  The ELRS
receiver apparently has both SPI and serial interfaces.  At least mine
works with a serial interface, I'm just ignoring the SPI port (if it is
really there).  Anyway, here are the files that make it work for me.  I
have a GPS hooked up to UART2, and though it doesn't have a barometer it
has flown OK (at least until I crashed the plane).

-Russ

mirov, 2024-04-22:
> I have been using this FC with INAV
>
> How do you do it?

That's a very broad question.  Have you tried already to put INAV onto that
FC and had trouble ?  If you are asking how to do that (I've supplied a tar
file with the target source files), you'll have to learn a lot more first...

donataso, 2024-05-11:
@mirov Could you please add the files you mentioned? They seem to not be attached to the comment.

I'm completely new with iNav, so it would greatly help me start with F411SX1280 configuration.

mirov, 2024-05-12:
If you are new to INAV, honestly I wouldn't use this controller (though I
have several myself).  The new version of INAV no longer supports the
STM32F411 processor that is used on this controller, so you'll have to use
an older 6.x INAV.   Also, this is a 1S only controller, and I've found
that for even a small fixed wing 1S just doesn't provide enough power (for
long enough anyway) to make it all work well.  2S is available on newer
versions of the controller and would probably make things much better
(though heavier).

Here's a 6.1.0 .hex file however, along with the target.c and target.h
files if you'd like to make any changes.  Depending on how you hook things
up you may need to play with it a bit.

-Russ

sensei-hacker, 2024-05-12:
As I recall, the plan is to drop F411 for INAV 8.0. So it should compile under 7.1.1.

We aren't accepting NEW F411 targets since the next version won't (necessarily) support them.

MrD-RC, 2024-05-12:
INAV 7.1 was the last release with new F411 targets. F411 is not good enough for a modern FPV setup. It is slow, has too little flash, and only 2 UARTs.

donataso, 2024-05-12:
@mirov You're probably attaching the files to the email reply, but they are not added to the comments:
https://github.blog/2011-03-10-reply-to-comments-from-email/#:~:text=Email%20attachments%20are%20currently%20ignored.

The drone I have is not exactly one above, but very similar: hglrc rekon3. It's HGLRCF411SX1280V2 on betaflight and should be quite similar, that's why I believe your config would be a great starting point for me. 

I specifically want to use this 1s drone as it's super light and my use case is taking it with me when hiking or backcountry skiing in the mountains for checking the path ahead. For this I need it to fly a bit more DJI-like instead of fpv. Adding gps and installing iNav should give it this ability.