iNavFlight/inav #7648: [4.0-rc1] Auto-CRUISE-3D?

ABLomas, 2021-12-06:
I had 4.0-rc1 on plane (i know that rc2 is available, but at this time i had only rc1). Was out on field and by accident put too much P on  pitch axis.
Take off, noticed oscillations, switched to manual, prepared to land, turned off acro/angle, overshoot, angle mode WOT, oscillations and:

- expected - some oscillations, i turn on manual, they stop, i do another circle, land and adjust PID's
- reality - airplane got into strange CRUISE-3D mode (at least blackbox says so), i had no controls on roll/pitch (at least, noticeable) while loosing altitude at the same time, added throttle to fly over road and ended up in forest (no damage to plane)

Now looking into blackbox (3% - sorry, no better resolution) i see that i have clear markers where flight mode has changed (angle, manual and so on), but no marker on last mode change (in blackbox log drama starts at 2:15). But on values table i see that plane got into CRUISE 3D mode. I have NAV_CRUISE on switch, but that switch not close to throttle/angle/attitude controls, i really, really doubt i could trigger it by accident.
Even if i triggered that CRUISE-3D mode - there are still questions:

- why it did not hold altitude? In DVR altitude acts strange (that mode is on from ~00:07 in video) - it was quickly loosing altitude, then reset, then again loosing alt (plane itself cruises at ~10% thr, it's overpowered wing and even unpowered glide ratio is way better than i see in video)
- why there is no mode change line in blackbox?

In my opinion - this cruise mode was triggered by some glitch after oscillations. Any comments?

[cruise_g_diff.txt](https://github.com/iNavFlight/inav/files/7658955/cruise_g_diff.txt)
[blackbox_log_2021-12-05_231159.zip](https://github.com/iNavFlight/inav/files/7658884/blackbox_log_2021-12-05_231159.zip)
Flight video (last seconds): https://youtu.be/K37tptYSjBs

MrD-RC, 2021-12-06:
1. Get the mode on your OSD. Otherwise you don’t know what’s supposed to be happening
2. NAV CRUISE mode is CRUISE-3D in the blackbox. I’m guessing it needs renaming. 
3. That tune is really bad. You can’t expect CRUISE to behave when it flies like that. 

It looks like you’ve set the pitch PID back to the defaults. Was the video you posted before setting the PID back to default? 
Did you Autotune in ACRO?

ABLomas, 2021-12-06:
I am trying to relate your answer to my question but cannot ;-)

Plane was **expected** to fly bad with such pitch P. It's OK, i could just land and adjust it (i did mistake just before flight, entered "15" instead of "5").
What is **unexpected** - CRUISE-3D, which is uncommanded (i did not turned it on) and which blocked all inputs except throttle, while loosing altitude at the same time.

ABLomas, 2021-12-06:
No, diff was provided straight after flight, nothing altered.
I am sure that i did not hit cruise switch, as it is far on my remote, not easily reacheable.
Again, let's assume i turned on this switch, this leaves two questions open:

- why i was loosing altitude in such big increments (blackbox shows pitch down attitude, ignoring stick value (pulling up)?
- why it is not shown on blackbox logs as mode change (everything else is shown on blackbox - from manual to acro, from acro to angle...)?

breadoven, 2021-12-06:
-ve pitch attitude is nose up not down, so it thought it was nose up, even though it was descending, which may have prevented it from passing a pitch up command to the servos because it thought it was already pitching up when in fact it wasn't. Smells of bad horizon drift which wouldn't be surprising given you have no Baro and instability causing excessive g-levels will upset gyro attitude estimates.

Not sure about the CRUISE_3D thing other than to say it's to be expected that it doesn't currently appear on the blackbox graph as a mode change, seems to have been missed. I use HEADINGHOLD and ALTHOLD together to obtain CRUISE rather than CRUISE mode itself. This is so I still have ALTHOLD if the GPS fails. If you use CRUISE by itself I'm pretty sure you lose both HEADINGHOLD and ALTHOLD if the GPS fails which isn't good, no fall back on altitude control. In my case when 3D CRUISE is active all I see on the blackbox graph is a mode change to ALTHOLD even though the blackbox tabular data shows 3D CRUISE in progress. Can't explain why 3D Cruise is triggering other than there's a glitch on the RC channels that's repeatedly switching it on and off. Does the other RC channel data ( rcData[x] ) make sense ... seems to jump around more than you'd expect ?

ABLomas, 2021-12-06:
Yeah, looks normal to me - no sudden changes, small oscillations about requested value, so at least first 4 channels looks fine.
I don't see other channels in blackbox explorer, they are not logged i guess (if they are - this would answer question about who triggered CRUISE). Link was close to perfect (ELRS 2.4G) all time, max range was maybe 150m at max distance and far from noisy environment, so radio glitches are unlikely. Yes, ELRS uses hybrid switches and AUX handling is not perfect, but at the same time - to trigger on/off cruise by radio glitch - it would require perfect timing to hit the same round-robin sequence (as hybrid switches are not transmitted at every frame).
I agree, that for flying far and away CRUISE should be set on OSD or combined with other modes. In my case - wing is 1m wingspan overpowered setup with super-fast heli servos, used to fly only around me, in close proximity, chase others and do acro, so i doubt i ever used CRUISE mode (and this is why it is mapped to switch which is not easily reacheable). What i have in close reach - 3-pos switch (MANUAL/ACRO/ANGLE) and this is more than enough for this wing.

Based on that mode change (16 times in 11 seconds) and GPS_GLITCH state (whatever it means) i still believe, that CRUISE-3D was triggered by some software glitch and not by RC command from ground.

ABLomas, 2021-12-06:
After some log checking on telegram with @stronnag and others - few things are cleared up:

- GPS_GLITCH are harmless (right?)
- navState changes in NAV_CRUISE mode happen in "manual-activated" flight, so many mode changes in flight - not indication of something wrong
- HDOP and other system params are +/- OK

So, back to initial state, this may be anything - my own hands doing something unexpected, TX glitch, uncommanded activation of NAV_CRUISE on INAV side. Without TX log or AUX channel logs on blackbox it's hard to tell.
Good thing that i missed busy road at approx 1m altitude, will remove this mode at all (in TX and INAV modes) and hope this won't repeat again ;-)

closed 2023-10-27 by DzikuVx