iNavFlight/inav #7659: OSD Characters Transposed on Matek F722-SE

jergling, 2021-12-15:
## Current Behavior
OSD displays with correct placement, but symbols are incorrect. For example, the GPS SATS icon appears as "^v", and the LQ icon appears as a "swap arrow". The numbers adjacent to these wrong or missing characters are often garbled as well, but general alphanumeric characters print fine (such as in system messages)

[image]

## Steps to Reproduce
1. Use Matek F722-SE
2. Install iNav v3.0.2
3. Configure as normal, _including updating fonts multiple times_
4. Power on device, view OSD

## Expected behavior
Symbols appear as demoed in the OSD pane of the configurator

[image]

## Suggested solution(s)
I'm not a dev and I apologize that I can't be more helpful. I originally thought this was some kind of memory offset problem, but there doesn't seem to be a fixed offset. 

## Additional context
This board works fine with Betaflight OSD, but I notice there's twice as many symbols in iNav
https://pastebin.com/YXnYPbtv

- FC Board name and vendor: MATEK F722-SE
- INAV version string: INAV/MATEKF722SE 3.0.2 Sep 15 2021 / 10:24:04 (66011184)
-GCC-9.3.1 20200408 (release)

Jetrell, 2021-12-15:
This is mention at the beginning of the 4.0 release notes.

**Font update required**
_A font file update is required to use the new symbols and avoid an invalid font warning. Upload the updated font of your choosing from the OSD tab._

MrD-RC, 2021-12-15:
You’re using the wrong configurator for the firmware. For 3.0.x use the 3.0.x configurator. At some point you must have used the 4.0.x configurator and uploaded the fonts using that.

closed 2021-12-15 by MrD-RC

jergling, 2021-12-15:
Thank you, I apologize for the confusion. I thought that configurator versions were independent from FW versions like Betaflight.