iNavFlight/inav #7665: No pitch & roll control below 1100us throttle in horizon mode / as expected following position hold on off cycle

dirtbag00, 2021-12-18:
## Current Behavior   FC Matek H743 slim  INAV 4.0.0 RC2
Flight controller seems to have no control of stability under 1100us throttle. The quad will just tip over and fall if you pull low throttle in horizon mode. 
If I switch to position hold then back into horizon mode the quad acts as expected, i.e It falls while maintaining pitch and roll.

## Steps to Reproduce
1. INAV 4.0.0 + Matek H743 slim
2. take off in horizon mode and pull throttle to zero.
2.quad loses control of pitch and roll 
3.
4.

## Expected behavior
Quad should maintain pitch and roll at any throttle while in horizon mode.

## Suggested solution(s)
It works as expected after engaging / disengaging position hold 

# version
# INAV/MATEKH743 4.0.0 Nov 26 2021 / 12:08:23 (f996170e)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor: Matek H743 slim
- INAV version string: 4.0.0 rc2

avsaase, 2021-12-18:
In the Configuration tab, enable the "Permanently enable airmode" toggle.

dirtbag00, 2021-12-18:
yea that's already on.

DzikuVx, 2021-12-19:
Airmode kicks in permanently only after Throttle crosses 1300us value. Just take off with more oomph of lower the `airmode_throttle_threshold` in CLI

closed 2021-12-19 by DzikuVx

dirtbag00, 2021-12-19:
Hey Pawel, thanks man! was struggling with that. Much appreciated