iNavFlight/inav #7672: FC Freeze randomly in Inav 4.0.0 with JetiEXBUS

RoadyFPV, 2021-12-19:
The FC Freezes randomly after a couple of minutes.

See here: https://github.com/iNavFlight/inav/issues/7434

RoadyFPV, 2022-01-01:
Would it be possible for a developer to compile the current release version with GCC 9 for me?

RoadyFPV, 2022-01-27:
After I have now tried a lot, I think I have found the issue why the FC freezes randomly.

It really seems to be with the ExBus protocol and the GCC10 compiler. Tests showed that the version with GCC9 compiler does not lead to any freezes. When I use INAV 4.1.0 rc1 with ExBus the FC freezes involuntarily.

Since the Jeti receivers can also output SBUS I switched to this protocol. The freezes are gone.
Unfortunately, the telemetry then no longer works.

This is a realy dangerous bug and should be followed up.

RoadyFPV, 2022-01-29:
@marv-t maybe u can assist here?

RoadyFPV, 2022-01-30:
I did some more testing today. I had connected the receiver to UART2 on the Matek f722se. There was always this freeze. I have now switched to UART6, there the problem does not seem to occur. I will continue to monitor and test this.

RoadyFPV, 2022-02-12:
@stronnag is it possible that you compile me 4.1 release with gcc 9?

stronnag, 2022-02-12:
no, I have no instances of old GCC.

RoadyFPV, 2022-02-19:
seems like when i disable telemetry feature the bug is gone

DzikuVx, 2022-05-10:
@RoadyFPV probably this fixes it https://github.com/iNavFlight/inav/pull/8024
I can not test thought

RoadyFPV, 2022-05-11:
@DzikuVx Thanks, i will test it asap.

RoadyFPV, 2022-05-11:
@DzikuVx I need a MATEKF722SE target to test it.

DzikuVx, 2022-05-12:
[inav_5.0.0_MATEKF722SE.zip](https://github.com/iNavFlight/inav/files/8675446/inav_5.0.0_MATEKF722SE.zip)
@RoadyFPV

RoadyFPV, 2022-05-12:
Thank you, Is this only a test version or is It flyable.

RoadyFPV, 2022-05-26:
Issue is still threre, fc freeze randomly

RobinatiNav, 2023-03-25:
Hi RoadyFPV, did the 50Hz change really help?

RoadyFPV, 2023-03-25:
Jea on inav i only have bench test....
Betaflight works without freezes on 50 hz

RobinatiNav, 2023-03-25:
> Jea on inav i only have bench test.... for flight i wait for inav 6 release and use gcc9
> 
> Betaflight works without freezes on 50 hz

Thanks. When you have iNav 6 compiled with GCC 9 could you share it with me please? I would like to test it.

RoadyFPV, 2023-03-25:
I am trying my best. Please remind me if I forget

RoadyFPV, 2023-03-25:
Wich Target do you need?

I compile inav 6 for myself MATEKF722SE with GCC9.3.1 right now, i hope it works.
This is a realy pain in the a....

RobinatiNav, 2023-03-25:
> Wich Target do you need?
> 
> I compile inav 6 for myself MATEKF722SE with GCC9.3.1 right now, i hope it works. This is a realy pain in the a....

MATEKH743

RobinatiNav, 2023-03-25:
> Wich Target do you need?
> 
> I compile inav 6 for myself MATEKF722SE with GCC9.3.1 right now, i hope it works. This is a realy pain in the a....

GOOD LUCK ;)

RoadyFPV, 2023-03-25:
[inav_6.0.0_MATEKH743GCC9.3.1.zip](https://github.com/iNavFlight/inav/files/11069835/inav_6.0.0_MATEKH743GCC9.3.1.zip)

RobinatiNav, 2023-03-25:
> [inav_6.0.0_MATEKH743GCC9.3.1.zip](https://github.com/iNavFlight/inav/files/11069835/inav_6.0.0_MATEKH743GCC9.3.1.zip)
> 
> Take care and give it a good bench test

WOW! Thank you very much :)

RoadyFPV, 2023-03-25:
your welcome

RobinatiNav, 2023-03-25:
> your welcome

I do not want to be bad guy but could you please compile this one too, please? MAMBAF722_WING I have 2 FCs. If you say No I am OK with it, You have done a lot already

RoadyFPV, 2023-03-25:
Pls test the first file first, and give me a feedback, i try it to compile it then.

RobinatiNav, 2023-03-25:
> Pls test the first file first, and give me a feedback, i try it to compile it then.

OK, deal. :) I have installed it already.

RoadyFPV, 2023-03-25:
pls share a cli.

-version and status

RobinatiNav, 2023-03-25:
> pls share a cli.
> 
> -version and status

I know I look stupid now. Could you please be more specific? I do not know what version and what status and what about cli you want. About cli do you mean DIFF?

RoadyFPV, 2023-03-25:
jea instead diff type version

RobinatiNav, 2023-03-25:
# version
# INAV/MATEKH743 6.0.0 Mar 25 2023 / 19:48:11 (GITDIR-N)
# GCC-9.3.1 20200408 (release)

RobinatiNav, 2023-03-25:
# status
INAV/MATEKH743 6.0.0 Mar 25 2023 / 19:48:11 (GITDIR-N)
GCC-9.3.1 20200408 (release)
System Uptime: 82 seconds
Current Time: 2041-06-28T03:04:00.000+02:00
Voltage: 15.30V (4S battery - OK)
CPU Clock=480MHz, GYRO=ICM42605, ACC=ICM42605, BARO=DPS310
STM32 system clocks:
  SYSCLK = 480 MHz
  HCLK   = 240 MHz
  PCLK1  = 120 MHz
  PCLK2  = 120 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x20020000, Heap available: 1716
I2C Errors: 50, config size: 9067, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = ADC 4, used = none
System load: 1, cycle time: 1003, PID rate: 997, RX rate: 124, System rate: 10
Arming disabled flags: CLI
OSD: MSP DisplayPort: DJI Compatability mode [30 x 16]
VTX: not detected

RoadyFPV, 2023-03-25:
thx looks good so far.. 
can u force the freeze?

RobinatiNav, 2023-03-25:
after a few min. I just lost the telemetry :(

RobinatiNav, 2023-03-25:
I changed the 100Hz to 50Hz now

RobinatiNav, 2023-03-25:
after about 5 min. I lost the telemetry again even with 50Hz :(

RoadyFPV, 2023-03-25:
only telemetry lost, fc is running?

i will test it tomorrow. my issue is telemetry is blinking and fc is frozen with a solid light an no reacton on usb

RobinatiNav, 2023-03-25:
> only telemetry lost, fc is running?
> 
> i will test it tomorrow. my issue is telemetry is blinking and fc is frozen with a solid light an no reacton on usb. Only with gcc10
> 
> i'm not a dev.

FC was OK and control surfaces were moving with a Tx sticks. Only telemetry was gone but I lost FC mainly during the flight.

RoadyFPV, 2023-03-25:
@marv-t can you take a look or can you guess what is going wrong?

RobinatiNav, 2023-03-25:
> @marv-t can you take a look or can you guess what is going wrong?

I do not know. I got frozen (I mean telemetry only) twice a couple minutes after it was flashed with hex file compiled in GCC 9.3.1 but after that it is still running without freezing. I was outside in a yard taxing on the grass and it was OK. It is windy outside this weekend. When it is calm down I will fly it.

RobinatiNav, 2023-03-25:
still ON and no freezing :)

RoadyFPV, 2023-03-26:
The main problem with gcc10 is that on bench looks ok but in mid flight the fc can freeze...
This cost me a Dart XL, the BUG is since using GCC10 an no one can find the BUG.
The reason why i compile it with gcc9 is that there is no freeze and save to fly.

RoadyFPV, 2023-03-26:
@RobinatiNav i have now flashd inav6 original with gcc10 and switch to 100 hz tx freq.
iam wiggeling the throttle stick as fast as i can and the fc is freezing instantly no surface wiggeling no telemetry

RoadyFPV, 2023-03-26:
the fc freezes after a few minutes without me doing anything
next test i switch to 50 hz

RobinatiNav, 2023-03-26:
> the fc freezes after a few minutes without me doing anything
> next test i switch to 50 hz

I am on 100Hz and 400Kz in I2C. So far no freezing but I have very minimalisting setup on Matek FC. There is only GPS module, LED strip with 2 LEDs (Arming, GPS), and EXBus connection to Rx. NO PWM outputs, no nav modes, no fail safe.

It was running over 1 hour yesterday with no freezing and it is running right now. If you send me .hex file for MAMBA it would be much better that one is fully installed.

RoadyFPV, 2023-03-26:
> > the fc freezes after a few minutes without me doing anything
> > next test i switch to 50 hz
> 
> I am on 100Hz and 400Kz in I2C. So far no freezing but I have very minimalistic setup on Matek FC. There is only GPS module, LED strip with 2 LEDs (Arming, GPS), and EXBus connection to Rx. No PWM outputs, no nav modes, no fail safe.
> 
> It was running over 1 hour yesterday with no freezing and it is running right now. If you send me .hex file for MAMBA it would be much better that one is fully installed.

Jea with GCC9 or?
Iam on stock inav6 with gcc10 right now

RobinatiNav, 2023-03-26:
> Jea with GCC9 or? Iam on stock inav6 with gcc10 right now

GCC 9.3.1 I am testing MATEK using GCC 9.3.1 too

RoadyFPV, 2023-03-26:
test with gcc10 50 HZ tx rate no freeze since around 30min test is runnin.

RobinatiNav, 2023-03-26:
> test with gcc10 50 HZ tx rate no freeze since around 30min test is runnin.

OK, I will try MAMBA with a stock 6.0 on 50Hz

RoadyFPV, 2023-03-26:
60 min no freeze with 50hz test still runnin. i can wiggle sicks telemetry ok....

RobinatiNav, 2023-03-26:
I am running it on 100Hz and receiver set from AUTO to half duplex ON. It is running over 10 min. with no freezing but I don't do any crazy throttle movement because my wife is at home and she doesn't like it. :D Well, she is on the way outside.... ;)

RobinatiNav, 2023-03-26:
I like Mike's idea: https://github.com/iNavFlight/inav/issues/8899#issuecomment-1483923863

RobinatiNav, 2023-03-26:
Wife is gone and after just a few second on throttle the telemetry got frozen.

RoadyFPV, 2023-03-26:
ok i will try to compile MAMBAF722_WING with gcc9

is target MAMBAF722_WING right?

RobinatiNav, 2023-03-26:
> ok i will try to compile MAMBAF722_WING with gcc9
> 
> is target MAMBAF722_WING right?

I think, it is a time.

RoadyFPV, 2023-03-26:
test with inav 6 and gcc10 on 50 hz 120min runtime no freeze and i cant force a freeze by wiggeling the sticks

RobinatiNav, 2023-03-26:
> test with inav 6 and gcc10 on 50 hz 120min runtime no freeze and i cant force a freeze by wiggeling the sticks

Maybe the problem is noise from ESC. I am really curious how it will go with .hex file from GCC 9.3.1

RoadyFPV, 2023-03-26:
i have 2x 60 amp esc connectet

RobinatiNav, 2023-03-26:
> i have 2x 60 amp esc connected

In MAMBA configuration, I have one 45A FMS predator V1 and in MATEK configuration I have 45A FMS Predator V2

RoadyFPV, 2023-03-26:
[inav_6.0.0_MAMBAF722_WINGGCC9.3.1.zip](https://github.com/iNavFlight/inav/files/11071262/inav_6.0.0_MAMBAF722_WINGGCC9.3.1.zip)

RobinatiNav, 2023-03-26:
> inav_6.0.0_MAMBAF722_WINGGCC9.3.1.zip

thank you :)

[22 more replies]

closed 2023-10-26 by DzikuVx