iNavFlight/inav #7674: SmartAudio not working on 4.0.0

andrzejroszkowski, 2021-12-20:
## Current Behavior
Matek F405-STD
Atlatl HV v2
SmartAudio connected to UART5 TX pin

No updates to Band/Channel/Power settings on VTX

## Expected behavior
Works with:
INAV/MATEKF405 3.0.2 Sep 15 2021 / 10:03:39 (66011184)
GCC-9.3.1 20200408 (release)

## Suggested solution(s)

## Additional context
https://pastebin.com/h6rFzSsB

---

- FC Board name and vendor: Matek F405 STD, Atlatl HV v2
- INAV version string:
INAV/MATEKF405 4.0.0 Dec 15 2021 / 14:53:05 (c20a9852)
GCC-10.2.1 20201103 (release)

Jetrell, 2021-12-24:
@andrzejroszkowski it maybe the _Atlatl HV v2_ ... I'm using smart audio with a _TBS Unify Pro 800mW_ on one of my quads. It works fine with 4.0.

andrzejroszkowski, 2021-12-24:
Works with 3.0.2 and BF 4.2

0crap, 2021-12-24:
Tested with a Matek F405-STD and a Rush TANK vtx, no issues.

```
# version
# INAV/MATEKF405 4.0.0 Dec 15 2021 / 14:53:05 (c20a9852)
# GCC-10.2.1 20201103 (release)
```

andrzejroszkowski, 2021-12-29:
INAV 3.0.2 - leds in first plane - blurry but one can see how band is switching
https://youtu.be/FD5wVHEZ3gc

andrzejroszkowski, 2021-12-29:
INAV 4.0.0 - as above, same setup (like 10 minutes earlier - just before flashing v3)
https://youtu.be/SwaSF2WM8ds

andrzejroszkowski, 2021-12-29:
Even more - tested with Atlatl HV micro and it works with 4.0, Holybro docs states same smartaudio version:
http://www.holybro.com/manual/Holybro_Atlatl_HV_micro_Manual_V1.1.pdf
http://www.holybro.com/manual/Holybro-Atlatl-HV-v2-Manual.pdf

closed 2022-05-14 by DzikuVx